Mixiaoxiao/Arduino-HomeKit-ESP8266 #93: Multiple bridges now working

ma7gdp, 2020-12-14:
I have 2 esp8266 devices (call them A and B) each configured as bridges (each with 3 accessories: temp, humidity and switch). Both bridges have unique name, serial and password. Each works when they are the first to be switched on. But the second one to be switched on does not respond. 

i.e. Switch A on then B and only A works. Switch B on then A and only B works.

Both work independently, so there is some kind of interaction between the two devices which is causing the issue. Any ideas? Many thanks.

ma7gdp, 2020-12-16:
Incase others have this issue, I thought I would post back what I found.... I was using the WiFiConnect.h library with captive portal to allow initial setup of the homekit device on wifi.  I found that there is some kind of conflict between the libraries. However if you initialise the homekit first (i.e. call homekit_setup()) and then start the wifi, all is OK.  Try to initialise the wifi before the homekit_setup() and it does not work. Not sure why, but happy it now works.

closed 2020-12-16 by ma7gdp

micahvdm, 2022-03-18:
Hi @ma7gdp would you mind sharing your code of how you set up the captive portal and got it working?