Mixiaoxiao/Arduino-HomeKit-ESP8266 #95: Sketch complexity vs Homekit timeouts

sircuri, 2020-12-15:
Currently I have a very minimal version of your library running on a Sonoff ESP 8266 Basic switch. That runs absolutely great. The problem is that if I add any more complexity to the inner loop Homekit pair steps start to time-out.

At this point I'm trying to find out what the exact timing requirements are for Homekit to perform a correct pairing. I can't seem to find the exact point where the iOS device is giving up.
The first 2 steps of the pairing are working fine. After step 2 the Homekit library forces a client reconnect that also works correctly. The admin pairing keys are exchanged, but it failes right after the first call to GetCategories. The client gets disconnected and my iPhone shows an error that it is unable to get information about the accessories.

I'm very well aware of the fact that the ESP8266 on a Sonoff is very limited in CPU power, but I realy want to understand this issue and hopefully work around it in future projects. Could it also be related to the discussion about the requirement of a RTSOS system on an ESP?

Mixiaoxiao, 2020-12-17:
Solutions:

1. Only run `arduino_homekit_loop()` in your `loop` while homekit is not paried.  
 ```
homekit_server_t *homekit_server = arduino_homekit_get_running_server();
if(homekit_server != nullptr && homekit_server->paired){
   // already paired, run all your codes
}else{
  // not paired, run arduino_homekit_loop() only 
}

 ```

2. Carefully check all your other codes in `loop`. 
   Change the codes with `delay(X)` or `while(1) {X}` (maybe in some libraries) to another logic with out any time-delay.

sircuri, 2020-12-17:
Thank you. That was what I was looking for. I was unable to find out if the server was paired or not. Thanks again!

adriankae, 2021-01-08:
> Solutions:
> 
> 1. Only run `arduino_homekit_loop()` in your `loop` while homekit is not paried.
> 
> ```
> homekit_server_t *homekit_server = arduino_homekit_get_running_server();
> if(homekit_server != nullptr && homekit_server->paired){
>   // already paired, run all your codes
> }else{
>  // not paired, run arduino_homekit_loop() only 
> }
> ```
> 
> 1. Carefully check all your other codes in `loop`.
>    Change the codes with `delay(X)` or `while(1) {X}` (maybe in some libraries) to another logic with out any time-delay.

Could you explain, why this is important? Thanks!

Mixiaoxiao, 2021-02-04:
The cpu of esp8266 is a little weak, so we should give all cpu for the HomeKit when not paired. Pairing is a very complex computation and we should give much cpu as possible to do this to avoid timeout.

closed 2021-03-30 by Mixiaoxiao