Mixiaoxiao/Arduino-HomeKit-ESP8266 #96: Arduino Examples naming issue

brencerddwr, 2020-12-15:
The *.ino file does not match the name of the containing folder.  This causes the examples to not be recognized as arduino sketches.

The legacy examples are fine, it is the newer examples.

DirkRussellG, 2020-12-15:
I don't that is so. Put the .c file and .h file in the same folder as sketch and try again.

brencerddwr, 2020-12-16:
Compare the file naming structure for the legacy examples and the newer examples.  You will see a clear difference.

In Arduino IDE, if you go into file->examples->homekit-esp8266 all you will see is the legacy examples.

brencerddwr, 2020-12-16:
I have submitted a pull request that will resolve this issue.

https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/pull/97

DirkRussellG, 2020-12-16:
I assumed you were referring to the examples here on gethub

Mixiaoxiao, 2020-12-17:
Merged.

closed 2020-12-17 by Mixiaoxiao