Mixiaoxiao/Arduino-HomeKit-ESP8266 #103: Device lost the connection with clients after few hours.

JoseLuis73, 2021-01-01:
Works well for a short period of time. After a few hours the device lost the connection with clients.

Any suggestion? 
Thank you

>>> [1755017] HomeKit: Got new client: local 192.168.0.32:5556, remote 192.168.0.40:57180
>>> [1755027] HomeKit: [Client 1073676900] Pair Verify Step 1/2
>>> [1755347] HomeKit: Free heap: 37400
>>> [1755365] HomeKit: Got new client: local 192.168.0.32:5556, remote 192.168.0.41:49674
>>> [1755375] HomeKit: [Client 1073678244] Pair Verify Step 1/2
>>> [1755696] HomeKit: Free heap: 35184
>>> [1755715] HomeKit: [Client 1073676900] Pair Verify Step 2/2
>>> [1755722] HomeKit: [Client 1073676900] Found pairing with 53796B55-2E30-4B05-ADF2-8D96BEEF5929
>>> [1755745] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [1756508] HomeKit: [Client 1073676900] Verification successful, secure session established
>>> [1756517] HomeKit: Free heap: 36080
>>> [1756536] HomeKit: [Client 1073678244] Pair Verify Step 2/2
>>> [1756542] HomeKit: [Client 1073678244] Found pairing with 53796B55-2E30-4B05-ADF2-8D96BEEF5929
>>> [1756565] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [1757329] HomeKit: [Client 1073678244] Verification successful, secure session established
>>> [1757337] HomeKit: Free heap: 36144
>>> [1757342] HomeKit: [Client 1073676900] Get Accessories
>>> [1757405] HomeKit: [Client 1073678244] Get Accessories
Free heap: 36776, HomeKit clients: 5
Free heap: 36832, HomeKit clients: 5
Free heap: 36824, HomeKit clients: 5

JoseLuis73, 2021-01-04:
>>> [  58317] HomeKit: [Client 1073683124] Pair Verify Step 1/2
>>> [  58638] HomeKit: Free heap: 40192
Free heap: 40240, HomeKit clients: 1
>>> [  58686] HomeKit: [Client 1073683124] Disconnected!
>>> [  58691] HomeKit: [Client 1073683124] Closing client connection
Free heap: 42736, HomeKit clients: 0

adriankae, 2021-01-08:
I have a similar problem. My device uses the connection on a regular bases. Like every 5 to 10 minutes. But it usually reconnects.

Any ideas?

Mixiaoxiao, 2021-03-30:
> Hi @JoseLuis73, after disconnecting does it reestablish the connection again?
> 
> I have a similar problem. My device loses the connection on a regular bases. Like every 5 to 10 minutes. But it usually reconnects.
> 
> Any ideas?

My device loses the connection on a regular bases. Like every 5 to 10 minutes. But it usually reconnects.

That is true.

WiFi on iOS devices will disconnect after a little time when they go sleep (screen off) to save battery. The TCP keepalive feature is used in this library to know weather iOS WiFi is alive or not.

jenspr, 2021-07-11:
I am suffering from similar. All works fine a while .. maybe hours. Then iOS does say "no connection" (or no response).
I can ping the ESP8266 so WIFI seems ok and connected. 
I don't know where to troubleshoot.. :-(

Anyone ideas.

macSE1990, 2021-07-29:
I had the same problems and found out that it was caused by my router (Fritzbox 7950) that was configured to auto-chose wifi channels to optimise performance when other routers in the house use the same or nearby channels.
After I chose a fixed channel for my wifi, the ESP8266 works fine without the reconnect to homekit problem.

jenspr, 2021-08-03:
Ahh.. @macSE1990, that sounds like a good hint. 
I have not experience any interruption for quite a while. 
But the fritzbox were set to AutoChannel... (I guess it does not change Channel for a while). I set to fixed channel and will monitor it for a while. 

anyone know how to change the code so it is tolerant to such wifi channel switches.. so the ESP recovers WLAN?

thncode, 2021-09-16:
> Ahh.. @macSE1990, that sounds like a good hint.
> I have not experience any interruption for quite a while.
> But the fritzbox were set to AutoChannel... (I guess it does not change Channel for a [...]

Did this workaround work for you? Have the same problem, but fix channel did not help…

ishii11, 2022-07-13:
I have the same problem and I found the way to solve it: Reconfigure your wifi router, set the bandwidth of 2.4 G wifi to 20MHz only (Not Auto 20/40MHz or 40MHz only). I have tried this configuration on several different wifi routers and it works! never disconnected again!

jenspr, 2022-07-13:
Isnt it the better option to change the source code to reconnect on disconnect?
I have occasions where I had a restart of router (power outtage) and the microcontroller did not establish connection again.

artyomLisovskij, 2022-11-29:
> I have the same problem and I found the way to solve it: Reconfigure your wifi router, set the bandwidth of 2.4 G wifi to 20MHz only (Not Auto 20/40MHz or 40MHz only). I have tried this configuration on several different wifi routers and it works! never disconnect again!

not working

artyomLisovskij, 2022-11-29:
> Isnt it the better option to change the source code to reconnect on disconnect? I have occasions where I had a restart of router (power outtage) and the microcontroller did not establish connection again.

agree

HanserBurlao, 2022-12-17:
Same Problem here :(

HanserBurlao, 2022-12-17:
I try to make a code to restar every hour to avoid lost connection (If you use the esp8266 to do something like open, close or restant it never lost connection), but I dont find the rigth codes.

HanserBurlao, 2022-12-17:
Restart* = reboot.

HanserBurlao, 2022-12-19:
I've tried to restart the wifi connection with a loop, but still the same problem.