Mixiaoxiao/Arduino-HomeKit-ESP8266 #109: Dynamic Hostname/Homekit name using ESP.getChipId()

danielhelmstedt, 2021-01-12:
I have been building an Occupancy sensor based off a capacitive reading through analogRead(A0). It implements custom characteristics to read the sensor value, and define a threshold. It also implements WifiManager and ArduinoOTA. My goal is to just have a firmware ready without having to customise thresholds, names etc for each board.

I've been successful so far except when it comes to getting Homekit to use the custom/dynamic name.

In my setup loop I have 
`
    char out[20];  
    sprintf(out, "PressureSensor-%X",ESP.getChipId());  
    const char * serial_str = out;
`

I can then call WifiManager and ArduinoOTA using serial_str for the AP SSID/OTA hostname.

In my_accessory.c I have `extern char * serial_str` to import the string from the main .ino, however, if I put `HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, serial_str),`, or even if I break out a 
`homekit_characteristic_t cha_serial = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, serial_str);`,
I get an error 

my_accessory.c:63:3: error: initializer element is not constant
   HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_sensor, .services = (homekit_service_t*[]) {
   ^
my_accessory.c:63:3: error: (near initialization for '(anonymous)')
exit status 1
initializer element is not constant

It's this constant that's tricking me. Other libraries (esp-homekit-devices to name one) call the serial value as `char * serial`. I notice however in this library's types.h that it is looking for a string `.value = HOMEKIT_STRING_(serial),`

nullfx, 2021-01-28:
most devices use the mac address in the host and link local addresses. feel free to use what i'd done on my fork of this in your implementation, I know this isn't probably the best / right way to implement it (you can pass in the name that includes the mac address), but it was how I wanted to do it.

comparison of the 2 repos here: https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/compare/master...nullfx:master

paullj1, 2021-03-06:
You can do this by declaring the char array in your includes, and setting it to somethings static, then overwriting it before you call setup with the config.  I do it that way to dynamically set the serial number.

paullj1, 2021-03-06:
For example, in my .c file:
`extern "C" char serial[16];`
...
`            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, serial),`

And in my .ino file:
`  sprintf(serial, "SN%X\0", ESP.getChipId());`

danielhelmstedt, 2021-03-15:
@paullj1 This is the method I was looking for! So clean and does exactly what I wanted, thank you! :)

closed 2021-03-15 by danielhelmstedt