Mixiaoxiao/Arduino-HomeKit-ESP8266 #110: soft WDT reset

Izac9, 2021-01-17:
Hi, i'm getting this watchdog problem when trying to move my stepper more than 1%
`0x40222858: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40100499: __digitalWrite(uint8_t, uint8_t) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring_digital.cpp line 86
0x40100470: delayMicroseconds(unsigned int) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 210
0x4021d716: Stepper_28BYJ_48::setOutput(int) at C:\Users\semte\Documents\Arduino\libraries\Stepper_28BYJ_48\src\Stepper_28BYJ_48.cpp line 43
0x40216e8e: UpdatePosition() at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\examples\window_covering/window_covering.ino line 170
0x40216f44: cha_target_position_setter(homekit_value_t) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\examples\window_covering/window_covering.ino line 122
0x40100913: umm_free_core(void*) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40216294: homekit_characteristic_ex_old_setter at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\accessories.c line 395
0x40202644: cJSON_GetObjectItem at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\cJSON.c line 1823
0x40217f3c: process_characteristics_update(cJSON const*, _client_context_t*) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 2183
0x40100bcc: malloc(size_t) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40100913: umm_free_core(void*) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100b97: free(void*) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40219ee0: homekit_server_on_update_characteristics(_client_context_t*, unsigned char const*, unsigned int) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 2402
0x4021b3d8: homekit_server_on_message_complete(http_parser*) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 2821
0x40204dd8: http_parser_execute at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\http_parser.c line 1928
0x40100bcc: malloc(size_t) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4021a119: homekit_client_process(_client_context_t*) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 2938
0x4021a3aa: homekit_server_process(homekit_server_t*) at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 3120
0x40224141: __delay(unsigned long) at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 57
0x4021b6e5: arduino_homekit_loop() at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\src\arduino_homekit_server.cpp line 3570
0x40216da6: homekit_loop() at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\examples\window_covering/window_covering.ino line 148
0x40216dbd: loop() at C:\Users\semte\Documents\Arduino\libraries\HomeKit-ESP8266\examples\window_covering/window_covering.ino line 107
0x40223c00: loop_wrapper() at C:\Users\semte\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197`

**here is a serial:**

` [ 104260] HomeKit: Got new client: local 192.168.0.52:5556, remote 192.168.0.185:54455
>>> [ 104301] HomeKit: [Client 1073682044] Pair Setup Step 1/3
>>> [ 113589] HomeKit: [Client 1073682044] Pair Setup Step 2/3
>>> [ 113600] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [ 122481] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [ 127026] HomeKit: [Client 1073682044] Pair Setup Step 3/3
>>> [ 127056] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [ 127809] HomeKit: Added pairing with 0061BCB1-CF92-47F1-85F2-E0FE88B1EE0D
>>> [ 127882] HomeKit: Free saved_preinit_pairing_context
>>> [ 127887] HomeKit: [Client 1073682044] Successfully paired
>>> [ 128104] HomeKit: [Client 1073682044] Disconnected!
>>> [ 128109] HomeKit: [Client 1073682044] Closing client connection
>>> [ 128327] HomeKit: Got new client: local 192.168.0.52:5556, remote 192.168.0.185:54456
>>> [ 128337] HomeKit: [Client 1073682020] Pair Verify Step 1/2
>>> [ 128659] HomeKit: Free heap: 42544
>>> [ 128846] HomeKit: [Client 1073682020] Pair Verify Step 2/2
>>> [ 128852] HomeKit: [Client 1073682020] Found pairing with 0061BCB1-CF92-47F1-85F2-E0FE88B1EE0D
>>> [ 128876] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [ 129639] HomeKit: [Client 1073682020] Verification successful, secure session established
>>> [ 129648] HomeKit: Free heap: 42656
>>> [ 129803] HomeKit: [Client 1073682020] Get Accessories
>>> [ 129892] HomeKit: [Client 1073682020] Update Characteristics
>>> [ 129933] HomeKit: [Client 1073682020] Update Characteristics
>>> [ 129974] HomeKit: [Client 1073682020] Update Characteristics
>>> [ 130016] HomeKit: [Client 1073682020] Get Characteristics
>>> [ 140407] HomeKit: [Client 1073682020] Update Characteristics
nahoru
0
0
>>> [ 140424] HomeKit: [Client 1073682020] Got characteristic 1.10 change event
>>> [ 140431] HomeKit: [Client 1073682020] Got characteristic 1.8 change event
nahoru
1
1
>>> [ 140448] HomeKit: [Client 1073682020] Got characteristic 1.10 change event
>>> [ 140455] HomeKit: [Client 1073682020] Got characteristic 1.8 change event
nahoru
2
1
.
.
.
[... 147 lines cut ...]

    }else if(current_position<pos){
      Serial.println("nahoru");
      Serial.println(current_position);
      Serial.println(position_state);
      small_stepper.step(1);
      current_position = current_position + 1;
      position_state = 1;
      homekit_characteristic_notify(&cha_position_state, cha_position_state.value);
      homekit_characteristic_notify(&cha_current_position, cha_current_position.value);
    }
    
    
    }
    position_state = 2;
    Serial.println(position_state);
    digitalWrite(D1, LOW);
    digitalWrite(D2, LOW);
    digitalWrite(D3, LOW);
    digitalWrite(D4, LOW);
    homekit_characteristic_notify(&cha_position_state, cha_position_state.value);
    homekit_characteristic_notify(&cha_current_position, cha_current_position.value);
  }
  `
Can you please help me where is the problem ?

closed 2021-01-25 by Izac9

edouardmenayde, 2021-03-21:
Hey, were you able to solve this ?

Izac9, 2021-03-21:
> Hey, were you able to solve this ?

Hi, I solved my problem. I created code where you don't use WHILE but instead of while you use IF.