Mixiaoxiao/Arduino-HomeKit-ESP8266 #112: Port update

awasspl, 2021-02-04:
Hi Mixiaoxiao,

I saw You're still active :) closing tickets.
Do You have any plans to update this port? I think Maxim Kulkin make a greate work with his esp8266 library and this Port is very important!! 
I love to use cpp with wire.h and be able to use You're homekit. 
Maxim update his  server library  couple of times since Your last commit so i was wondering if it's a good idea to start with You're work becouse if You don't want to work on it any more this super extra cool project will be dead :(

Mixiaoxiao, 2021-03-30:
Sorry, no time to maintain. I think current version is stable enough.

awasspl, 2021-04-16:
Not exactly. 

Main fork was update with many bugs fixed. 
For example Your's port still is exposed for remote reset.
curl -X POST http://DEV_IP:5556/pair-setup --data a

So your version is greate but not suitable for use :(