Mixiaoxiao/Arduino-HomeKit-ESP8266 #115: Help with temperature example

gio-dude, 2021-02-15:
I'm trying to add the DHT22 sensor. I was able to get temperature working but I'm having issue with humidity. I've new to programing and still surprised I was able to get temperature to work. I've included my modifications to the example code. Any guidance would be much appropriate. I get errors at  cha_humidity.value.float_value = humidity_value;

`/*
 * temperature_sensor.ino
 *
 * This example shows how to:
 * 1. define a temperature sensor accessory and its characteristics (in my_accessory.c).
 * 2. report the sensor value to HomeKit (just random value here, you need to change it to your real sensor value).
 *
 *  Created on: 2020-05-15
 *      Author: Mixiaoxiao (Wang Bin)
 *
 * Note:
 *
 * You are recommended to read the Apple's HAP doc before using this library.
 * https://developer.apple.com/support/homekit-accessory-protocol/
 *
 * This HomeKit library is mostly written in C,
 * you can define your accessory/service/characteristic in a .c file,
 * since the library provides convenient Macro (C only, CPP can not compile) to do this.
 * But it is possible to do this in .cpp or .ino (just not so conveniently), do it yourself if you like.
 * Check out homekit/characteristics.h and use the Macro provided to define your accessory.
 *
 * Generally, the Arduino libraries (e.g. sensors, ws2812) are written in cpp,
 * you can include and use them in a .ino or a .cpp file (but can NOT in .c).
 * A .ino is a .cpp indeed.
 *
 * You can define some variables in a .c file, e.g. int my_value = 1;,
 * and you can access this variable in a .ino or a .cpp by writing extern "C" int my_value;.
 *
 * So, if you want use this HomeKit library and other Arduino Libraries together,
 * 1. define your HomeKit accessory/service/characteristic in a .c file
 * 2. in your .ino, include some Arduino Libraries and you can use them normally
 *                  write extern "C" homekit_characteristic_t xxxx; to access the characteristic defined in your .c file
 *                  write your logic code (eg. read sensors) and
 *                  report your data by writing your_characteristic.value.xxxx_value = some_data; homekit_characteristic_notify(..., ...)
 * done.
 */

#include <Arduino.h>
#include <arduino_homekit_server.h>
#include "wifi_info.h"
#include "DHT.h"
//include the Arduino library for your real sensor here, e.g. <DHT.h>

#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

#define DHTPIN 2 
#define DHTTYPE DHT22 
DHT dht(DHTPIN, DHTTYPE);

void setup() {
	Serial.begin(115200);
	wifi_connect(); // in wifi_info.h
	my_homekit_setup();

 dht.begin();
}

void loop() {
	my_homekit_loop();
	delay(10);
}

//==============================
// Homekit setup and loop
//==============================

// access your homekit characteristics defined in my_accessory.c
extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_current_temperature;

static uint32_t next_heap_millis = 0;
static uint32_t next_report_millis = 0;

void my_homekit_setup() {
	arduino_homekit_setup(&config);
}

void my_homekit_loop() {
	arduino_homekit_loop();
	const uint32_t t = millis();
	if (t > next_report_millis) {
		// report sensor values every 10 seconds
		next_report_millis = t + 10 * 1000;
		my_homekit_report();
	}
	if (t > next_heap_millis) {
		// show heap info every 5 seconds
		next_heap_millis = t + 5 * 1000;
		LOG_D("Free heap: %d, HomeKit clients: %d",
				ESP.getFreeHeap(), arduino_homekit_connected_clients_count());

	}
}

void my_homekit_report() {
	float temperature_value = dht.readTemperature(true); // FIXME, read your real sensor here.
	cha_current_temperature.value.float_value = temperature_value;
	LOG_D("Current temperature: %.1f", temperature_value);
	homekit_characteristic_notify(&cha_current_temperature, cha_current_temperature.value);
 
  float humidity_value = dht.readHumidity(); // FIXME, read your real sensor here.
  cha_humidity.value.float_value = humidity_value;
  LOG_D("Current humidity: %.1f", humidity_value);
  homekit_characteristic_notify(&cha_humidity, cha_humidity.value);
}

int random_value(int min, int max) {
	return min + random(max - min);
}`

Varogkorn, 2021-02-17:
You can get some idea from my work. [DHT_Switch.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/5993334/DHT_Switch.zip) I have 2 LED and Report Temperature with Humidity from DHT22.

gio-dude, 2021-02-17:
> You can get some idea from my work. [DHT_Switch.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/5993334/DHT_Switch.zip) I have 2 LED and Report Temperature with Humidity from DHT22.

Thank you so much after looking at your code I left out the extern "C" homekit_characteristic_t cha_humidity; line

closed 2021-02-17 by gio-dude

alexandrerochasalles, 2021-04-01:
> You can get some idea from my work. [DHT_Switch.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/5993334/DHT_Switch.zip) I have 2 LED and Report Temperature with Humidity from DHT22.

This have also solved my issue, thanks a lot.

Just let me know something, how should I know that in the accessory.c the HomeKit_characteristic_t cha_humidity should be (CURRENT_RELATIVE_HUMIDITY) ???
Because I was trying for a lot without the "relative" in the middle, and always getting an error...

Varogkorn, 2021-04-04:
> > You can get some idea from my work. [DHT_Switch.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/5993334/DHT_Switch.zip) I have 2 LED and Report Temperature with Humidity from DHT22.
> 
> This have also solved my issue, thanks a lot.
> 
> Just let me know something, how should I know that in the accessory.c the HomeKit_characteristic_t cha_humidity should be (CURRENT_RELATIVE_HUMIDITY) ???
> Because I was trying for a while without the "relative" in the middle, and always getting an error...

- Hi,@alexandrerochasalles It is a required characteristics for homekit. If you not include the relative. it is not gonna think it is not in a `HOMEKIT_SERVICE_HUMIDITY_SENSOR` as in the picture.

[image]