Mixiaoxiao/Arduino-HomeKit-ESP8266 #116: DHT22 sensor error at start up

gio-dude, 2021-02-18:
So I don't know if this hardware or software issue. I got the example temperature sketch to work but after start up the serial monitoring display nam for the results of the temp and Humidity until i disconnect power pin to the sensor and reconnect it. then it works fine. 

Im using a nodeMCU I have the sensor connected to pin D4 with a 10k resistor between data and power.

Varogkorn, 2021-03-06:
@gio-dude I have the same problem same as you with DHT22. I didn't find what's the problem. But I just fix the problem by using some trasistor to control 3.3V. When it boot finish quite a while I program it to reboot to turn off and on DHT22.

gio-dude, 2021-03-06:
Ok so if I understand correctly you are using transistors to power cycles he temperature sensor once the esp finishes its boot cycle.

Varogkorn, 2021-03-08:
> Ok so if I understand correctly you are using transistors to power cycles he temperature sensor once the esp finishes its boot cycle.

Yes