Mixiaoxiao/Arduino-HomeKit-ESP8266 #118: Switching to other device

tk97tk, 2021-02-23:
I wanted to switch the ui from a usual lightbulb to a TV ui. So i have changed the the **.category** of the **HOMEKIT_ACCESSORY** from **homekit_accessory_category_lightbulb** to **homekit_accessory_category_television**.

Now in my home app, it shows the tv UI and I can activate the device through a long press and a repeat press. Everything works fine. The TV turns on. 
Unfortunately, the button says updating all the time and I can not do a short tap to activate the device. 
Where I have to change something, to remove the updating?

[image]

[image]
[image]
[image]

ApplebaumIan, 2021-02-23:
How are you checking whether the device is on or off?

tk97tk, 2021-02-23:
I have used the default project simplest_led for testing

ApplebaumIan, 2021-02-24:
So I think what's going on is that Homekit needs more than just on or off for a TV characteristic state. It could be something like "playing" or "not playing". I can't seem to find tv in Apple's HAP protocol docs.

tk97tk, 2021-02-24:
The missing part is the port for choosing between the inputs (HMDI ...) for the tv, which I don't need 😅. The perfect solution would be a dummy function.

awasspl, 2021-02-24:
Regarding HAP 
TV Category need television_service not lightbulb see You code. You mix accessory. HK get information that it's a TV accessory and you send light services with ON/off charactersitic.

Update definion:
1) TV category
2) TV service (not lightbulb)
3)  Required Characteristics for TV service (you have to add and updates them all to work properly)
 - ACTIVE
 - ACTIVE_IDENTIFIER
 - CONFIGURED_NAME
 - SLEEP_DISCOVERY_MODE
 
Optional:
 - IDENTIFIER
 - INPUT_DEVICE_TYPE
 - TARGET_VISIBILITY_STATE

ApplebaumIan, 2021-02-24:
I also can't seem to recreate the issue. My ESP8266 shows up as a lightbulb. lol

tk97tk, 2021-02-24:
@ApplebaumIan Yeah, you have to add television instead of lightbulb in the HOMEKIT_SERVICE. 

@awasspl  I also found this in the characteristics.h. But I have no clue how to initialized it probably. 
There is at least no compile error.
I have added them to the HOMEKIT_SERVICE.

[image]
[image]

awasspl, 2021-02-25:
First remove led_on characteristic becouse it's not support is TV servise
next You need to add all Required:

HOMEKIT_CHARACTERISTIC_(ACTIVE,0)
HOMEKIT_CHARACTERISTIC_(ACTIVE_IDENTIFIER,"")

tk97tk, 2021-02-27:
Yeah, it's working, I added the methods to query the state of the homekit_format_uint8. I can now control my tv.

closed 2021-02-28 by tk97tk

hitenlulla, 2022-01-01:
@tk97tk Hello bro, Can you share your code? I'm trying to do it but it doesn't work. It'll be a great help. Thank you.