Mixiaoxiao/Arduino-HomeKit-ESP8266 #122: When restarting the device homekit always needs to be paired again

sEnesCinar, 2021-03-09:
Hello,

I flashed the switch code to my sonoff with esp8266 in it. Its working properly but when i unplug and replug the sonoff. It is turning to unavailable in homekit and it needs to be pair again. How can i fix that. Thank you.

sircuri, 2021-03-13:
Maybe post some relevant code so we can have a look.

sEnesCinar, 2021-03-13:
Thanks for the reply. Here is my main code and accessory.c files

[homekit.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/6135141/homekit.zip)

sircuri, 2021-03-13:
Well, look at your setup code. As I expected:

`homekit_storage_reset(); // to remove the previous HomeKit pairing storage when you first run this new HomeKit example`

sEnesCinar, 2021-03-13:
I tried to comment and uncomment that line but nothing changed.

sircuri, 2021-03-14:
That line needs to stay commented / deleted from your code.
In my project I moved that code to a long-press routine on the Sonoff. If you long press the button, the reset is called and all is wiped.

sEnesCinar, 2021-03-14:
It is a good idea i think. Can you share your code with me?

closed 2021-03-30 by sEnesCinar