Mixiaoxiao/Arduino-HomeKit-ESP8266 #128: You cannot add two identical devices

askarkurymbayev, 2021-04-19:
Greetings friends, faced with the problem of inability to add multiple devices with the same firmware, have any ideas about how to solve the problem?

Mixiaoxiao, 2021-04-20:
Gen the accessory name with MAC or chip-id.

askarkurymbayev, 2021-04-20:
you can help me generate a name with a mac address, I try with WiFi.macAddress (), but it doesn't work

jenspr, 2021-07-11:
Hi there, did you solve that.. I might have run into same problem.
I setup one instance on one ESP8266 and then I tried to build another one and I can't connect via HomeKit.. I get "Something went wrong" during pairing in iOS Home (after entering the pairing code 11111...).

I did not know if and what I should change in code? Any hint?

dsbaha, 2021-07-20:
I, too, have been looking at this.  I suspect you have to change it in your config definition.  Trying to find a way to generate a device name and a pairing password based on ESP.getChipID() or something.

askarkurymbayev, 2021-07-22:
good time of the day friends, who managed to solve the problem?
as I understand it, you need to somehow edit the line
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "xxxxxx");

dennisc1213, 2021-08-19:
> good time of the day friends, who managed to solve the problem?
> as I understand it, you need to somehow edit the line
> homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "xxxxxx");

I am a beginner to this. But I solved this issue by manually editing this line in My_accessory.c
` HOMEKIT_CHARACTERISTIC(NAME, "Multiple Sensors")` 

By editing "Multiple Sensors" to Unique words, I managed to run three identical devices (Temperature sensor) at home.

hitenlulla, 2021-12-20:
@dennisc1213 Hello, I did this and still I'm facing the issue. Can you help me out?

hitenlulla, 2021-12-20:
@jenspr @askarkurymbayev @dsbaha  Did you solve the issue? How did you generate accessory name in my_accessory.c? Can someone share the code?

madmacks59, 2021-12-31:
The specification states that device ID number "identifier" for accessories must be a unique random number generated whenever a factory reset happens (meaning when you build your code and deploy it to the ESP8266) and must persist across reboots. So, as stated before, make sure your identifiers are unique across your LAN...

hitenlulla, 2022-01-01:
@madmacks59 Thank you it worked, One more thing, how can we generate random number in the accessory.c file? Currently I have to hardcode a random number . Can you share your snippet where you generate the random number in the accessory.c file? Thankyou.