Mixiaoxiao/Arduino-HomeKit-ESP8266 #129: Device incompatible

andyKien, 2021-04-19:
Hi,

i basically used your temperature sensor example, but replaced it by an ambient light sensor.
The device is showing up in the home app, but when I try to connect it says the device is incompatible.
(I even tried with your exact example)

Im using a Node MCU V2 (ESP8266MOD)

Is there any known compatibility issue with this device?

best regards

Mixiaoxiao, 2021-04-20:
Accessory defination is invalid, or value is out of range. Check HAP doc.

andyKien, 2021-04-20:
Hi,

thanks for the response. 
After showing multiple different errors it randomly startet to be able to connect and also transmit data.

Unfortunately it is recognized as a light, instead of a sensor, despite setting the category to sensor ".category=homekit_accessory_category_sensor".

closed 2021-04-20 by andyKien