Mixiaoxiao/Arduino-HomeKit-ESP8266 #133: unable to add accessory

RudraR20, 2021-06-24:
hi sir this project is really awesome but i am having some error . actually i can the see the esp in my homekit app but as soon as i try to log into it its says unable to add accessory . sir please help me . and i am attaching some file please check it 

>[image]
  
>[image] 

>https://user-images.githubusercontent.com/81382826/123260430-2c582080-d513-11eb-9962-4d2c54aa52c1.mp4

 please reply as soon as possible

hassbian-ABC, 2021-06-25:
IOS15?

RudraR20, 2021-06-26:
ya

hassbian-ABC, 2021-06-26:
The problem with IOS 15 is very good on IOS 14

moonshot87, 2021-06-29:
Up. Having same problem

moonshot87, 2021-06-30:
> The problem with IOS 15 is very good on IOS 14

Any workaround?

RudraR20, 2021-06-30:
try to do on a raspberry pi

RudraR20, 2021-06-30:
it is much simpler

RudraR20, 2021-06-30:
and if you wanna do with esp with homekit i have a easy solution

RudraR20, 2021-06-30:
first of all download home assistant on any platform like windows , mac , raspberry pi . Then compile the this particular code to your esp module . Then download homekit controller on home assistant .  then download home kit on home assistant and scan . you will get you result

RudraR20, 2021-06-30:
if any problem contact - rudrarajputsingh1040@gmail.com

hassbian-ABC, 2021-07-01:
> first of all download home assistant on any platform like windows , mac , raspberry pi . Then compile the this particular code to your esp module . Then download homekit controller on home assistant . then download home kit on home assistant and scan . you will get you result

Oh no, using Arduino-Homekit-ESP8266 is to simplify the problem. You make the problem more complicated

RudraR20, 2021-07-01:
you think so

RudraR20, 2021-07-01:
i AM just 14 so if i can do then maybe you can do that too

RudraR20, 2021-07-01:
give it a try

RudraR20, 2021-07-01:
and I recommend to use a raspberry pi it is much much much more easy and can use 16 channel relay

hassbian-ABC, 2021-07-01:
> give it a try

No, Homeassistant is a cluttered platform that I gave up two years ago

moonshot87, 2021-07-01:
Do you have any sample of working code?

moonshot87, 2021-07-01:
> give it a try 
> 
> 

Do you have any sample of working code?

hassbian-ABC, 2021-07-01:
I prefer to use homebridge

RudraR20, 2021-07-01:
> > give it a try
> 
> No, Homeassistant is a cluttered platform that I gave up two years ago

THEN use homebridge

RudraR20, 2021-07-01:
> I prefer to use homebridge

you can use

RudraR20, 2021-07-01:
> > give it a try
> 
> Do you have any sample of working code?

this is too simple

RudraR20, 2021-07-01:
do you guys have a raspberry pi ??
any model will work

RudraR20, 2021-07-01:
its simple

RudraR20, 2021-07-01:
and guys i have my own voice assistant named  jarvis

hassbian-ABC, 2021-07-01:
This is a bug on IOS 15. There's no need to do anything unnecessary

RudraR20, 2021-07-01:
broo i have 3 iphone

RudraR20, 2021-07-01:
one with ios 15 and other with ios 14

RudraR20, 2021-07-01:
i have tried

RudraR20, 2021-07-01:
and i am just recommending the other way with is more easy according to me

RudraR20, 2021-07-01:
where are you from ??

closed 2021-07-01 by RudraR20

PR77, 2022-08-22:
> This is a bug on IOS 15. There's no need to do anything unnecessary

Is this really an IOS 15 bug or has something changed where the pairing times are enforced? I made several homekit devices and recently I had to remove one device and re-pair but this did not work unfortunately. After a lot of research, it seems IOS 15.x introduced a forced pairing time which the ESP8266 can't achieve.

devangs3, 2022-10-22:
Reporting same issue on iOS 16.1 beta 
[image]
[image]
[image]
[image]
[image]