Mixiaoxiao/Arduino-HomeKit-ESP8266 #136: Adding new user to home drops paired devices

paullj1, 2021-07-05:
When adding a user to a home, I’m guessing a new ID should be generated, or pairing might need to be renegotiated in some way.  As is, when adding a new user to the home, devices built with this library are no longer reachable, and must be re-paired as new devices.

paullj1, 2021-07-05:
Thinking this is a bug in this library, since other HomeKit devices are fine.

SoprachevAK, 2021-07-20:
I create a pair on one device, and add the other accounts to the home application, which immediately have access to control and everything works. However, I have Apple TV as a hub.

dsbaha, 2021-07-22:
I was kinda having a similiar issue until I put in yield statements, see [PR 138](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/pull/138)

paullj1, 2021-07-22:
Yeah, I also have a few Apple TV’s and HomePods.  I don’t think it’s a hub thing.  I’m afraid to do any more testing since I have about 20 devices that use this library, and adding them all back in, and setting up all my automations/scenes again was a huge pain.  Curious to see if the PR gets merged though.