Mixiaoxiao/Arduino-HomeKit-ESP8266 #137: BME280 Causing issues

DannyAt, 2021-07-09:
I get this error when I add code to read temp off the BME280

>>> [ 253349] HomeKit: Init server over

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021d0d0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 0190
3fffff40:  00000000 00000000 4bc6a7f0 00000000  
3fffff50:  3fffdad0 3ffef83c 401004e9 00000000  
3fffff60:  0003dda8 00000000 3ffef65c 4021d1b1  
3fffff70:  00000000 00000000 0003dda8 40216c77  
3fffff80:  3fffdad0 00000000 0003dda8 40216cd7  
3fffff90:  3fffdad0 00000000 3ffefa34 40216d0d  
3fffffa0:  feefeffe feefeffe feefeffe 402238d8  
3fffffb0:  feefeffe feefeffe 3ffe8a14 40100f5d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00074a00
~ld
WiFi connecting...

Any Ideas?

PR77, 2022-08-14:
I have a project which is using the BME280 without issue... Are you still having problems or have they been resolved?

DannyAt, 2022-08-14:
I got it working but the temp isn’t reading correctly, it’s reading about 4 degrees higher.

ZhuLinzhu, 2022-08-22:
I have same problem with u, and i have no idea about how to slove it.