Mixiaoxiao/Arduino-HomeKit-ESP8266 #141: Could not complete opoeration

pratikdesai189, 2021-08-10:
Hello I used your code and modified it for 8 Switches.I am able to add 7 switches but at the time of adding 8th switch it is giving me error  **" Could not complete opoeration "** when i click on OK it is saying to exit and after that I can see only 7 Switches in Homekit app.Please help to solve the issue.
@Mixiaoxiao @jmbcpi 
>  " Could not complete opoeration "

pratikdesai189, 2021-08-16:
> Hello I used your code and modified it for 8 Switches.I am able to add 7 switches but at the time of adding 8th switch it is giving me error **" Could not complete operation "** when i click on OK it is saying to exit and after that I can see only 7 Switches in Homekit app.Please help to solve the issue.
> @Mixiaoxiao @jmbcpi
> 
> > " Could not complete operation "

pratikdesai189, 2021-08-16:
Hello I used your code and modified it for 8 Switches.I am able to add 7 switches but at the time of adding 8th switch it is giving me error " Could not complete operation " when i click on OK it is saying to exit and after that I can see only 7 Switches in Homekit app out of that the 1st switch and 7th switch not working.Please help to solve the issue.
@Mixiaoxiao @jmbcpi

jmbwell, 2021-08-16:
I'm not sure how exactly to help resolve it. I will say when I've gotten that message, it has been either because the key exchange took too long and HomeKit timed out, or because the ESP device ran out of memory and reset, preventing the association from completing — either way, asking too much of the little ESP chip. I would suggest adding some debug logging to your code and running it with the serial monitor enabled, so that you can watch the output for clues. You might also consider revising your approach to use fewer HomeKit devices per ESP board. 

It's a major feat to get the ESP8266 working with this at all. It's very easy to exceed its capacity. Good luck.

pratikdesai189, 2021-08-17:
@jmbwell Thank You for replying I am using ESP32 so memory is not issue I think.I will try your key exchange solution by adding debug logging.The issue is 7 devices are working fine,But when i modified code for for 8th device its not adding when I modified code for 9 device the 9th one not adding and so on.