Mixiaoxiao/Arduino-HomeKit-ESP8266 #142: Latest version doesn't appear in PlatformIO Registry

jomiplaz, 2021-08-18:
Only appears up to version 1.2.0

[https://platformio.org/lib/show/7045/HomeKit-ESP8266/installation](url)

Timwun, 2021-09-11:
Still onyl 1.2.0 on PlatformIO... :/

ajschwieterman, 2022-12-04:
Could it be because of these files?

https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/blob/8a8e1a065005e9252d728b24f96f6d0b29993f67/library.json#L15

https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/blob/8a8e1a065005e9252d728b24f96f6d0b29993f67/library.properties#L2

v1.2.0 is the latest version that shows in the Arduino IDE, too.
[image]