Mixiaoxiao/Arduino-HomeKit-ESP8266 #144: the heap is decreasing

AGRO666, 2021-08-25:
When HomeKit clients start to disconnect, the heap decreases. After several client disconnections, the heap becomes insufficient, and the ESP8266 is rebooted.

`11:03:15.249 -> HomeKit reporting
11:03:15.249 -> >>> [ 211285] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:03:15.249 -> >>> [ 211300] HomeKit: [Client 1073696988] Got characteristic 1.13 change event
11:03:15.249 -> >>> [ 211307] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:03:15.249 -> >>> [ 211314] HomeKit: [Client 1073696988] Got characteristic 1.16 change event
11:03:15.249 -> >>> [ 211322] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:03:15.249 -> >>> [ 211329] HomeKit: [Client 1073696988] Got characteristic 1.20 change event
11:03:15.249 -> >>> [ 211336] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:03:15.249 -> >>> [ 211343] HomeKit: [Client 1073696988] Got characteristic 1.19 change event
11:03:15.249 -> >>> [ 211350] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:03:15.249 -> >>> [ 211357] HomeKit: [Client 1073696988] Got characteristic 1.10 change event
11:03:21.818 -> >>> [ 217946] HomeKit: [Client 1073696988] Get Characteristics
11:03:24.736 -> HTTP connected to Drivent 192.168.1.139
11:03:25.242 -> Temp:23.30, Hum:36, Pres:768, CO2:740, SW:1
11:03:25.242 -> Free heap:22352, HomeKit client:2
11:03:25.242 -> HomeKit reporting
11:03:25.242 -> >>> [ 221281] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:03:25.242 -> >>> [ 221296] HomeKit: [Client 1073696988] Got characteristic 1.13 change event
11:03:25.242 -> >>> [ 221303] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:03:25.242 -> >>> [ 221310] HomeKit: [Client 1073696988] Got characteristic 1.16 change event
11:03:25.242 -> >>> [ 221317] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:03:25.242 -> >>> [ 221324] HomeKit: [Client 1073696988] Got characteristic 1.20 change event
11:03:25.242 -> >>> [ 221331] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:03:25.242 -> >>> [ 221339] HomeKit: [Client 1073696988] Got characteristic 1.19 change event
11:03:25.242 -> >>> [ 221346] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:03:25.242 -> >>> [ 221353] HomeKit: [Client 1073696988] Got characteristic 1.10 change event
11:03:28.635 -> >>> [ 224760] HomeKit: [Client 1073697372] Get Characteristics
11:03:34.768 -> HTTP connected to Drivent 192.168.1.139
11:03:35.204 -> Temp:23.30, Hum:35, Pres:768, CO2:737, SW:1
11:03:35.204 -> Free heap:20144, HomeKit client:2
11:03:35.204 -> HomeKit reporting
11:03:35.204 -> >>> [ 231230] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:03:35.204 -> >>> [ 231245] HomeKit: [Client 1073696988] Got characteristic 1.13 change event
11:03:35.204 -> >>> [ 231252] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:03:35.204 -> >>> [ 231259] HomeKit: [Client 1073696988] Got characteristic 1.16 change event
11:03:35.204 -> >>> [ 231266] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:03:35.204 -> >>> [ 231273] HomeKit: [Client 1073696988] Got characteristic 1.20 change event
11:03:35.204 -> >>> [ 231280] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:03:35.204 -> >>> [ 231287] HomeKit: [Client 1073696988] Got characteristic 1.19 change event
11:03:35.204 -> >>> [ 231294] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:03:35.204 -> >>> [ 231301] HomeKit: [Client 1073696988] Got characteristic 1.10 change event
11:03:44.738 -> HTTP connected to Drivent 192.168.1.139
11:03:45.176 -> Temp:23.30, Hum:35, Pres:768, CO2:734, SW:1
11:03:45.244 -> Free heap:16736, HomeKit client:2
11:03:45.244 -> HomeKit reporting
11:03:45.244 -> >>> [ 241279] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:03:45.244 -> >>> [ 241294] HomeKit: [Client 1073696988] Got characteristic 1.13 change event
11:03:45.244 -> >>> [ 241301] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:03:45.244 -> >>> [ 241308] HomeKit: [Client 1073696988] Got characteristic 1.16 change event
11:03:45.244 -> >>> [ 241315] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:03:45.244 -> >>> [ 241322] HomeKit: [Client 1073696988] Got characteristic 1.20 change event
11:03:45.244 -> >>> [ 241329] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:03:45.244 -> >>> [ 241336] HomeKit: [Client 1073696988] Got characteristic 1.19 change event
11:03:45.244 -> >>> [ 241343] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:03:45.244 -> >>> [ 241350] HomeKit: [Client 1073696988] Got characteristic 1.10 change event
11:03:54.732 -> HTTP connected to Drivent 192.168.1.139
11:03:55.269 -> Temp:23.30, Hum:36, Pres:768, CO2:732, SW:1
11:03:55.269 -> Free heap:15328, HomeKit client:2
11:03:55.269 -> HomeKit reporting
11:03:55.269 -> >>> [ 251303] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:03:55.269 -> >>> [ 251318] HomeKit: [Client 1073696988] Got characteristic 1.13 change event
11:03:55.269 -> >>> [ 251325] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:03:55.269 -> >>> [ 251332] HomeKit: [Client 1073696988] Got characteristic 1.16 change event
11:03:55.269 -> >>> [ 251340] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:03:55.269 -> >>> [ 251347] HomeKit: [Client 1073696988] Got characteristic 1.20 change event
11:03:55.269 -> >>> [ 251354] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:03:55.269 -> >>> [ 251361] HomeKit: [Client 1073696988] Got characteristic 1.19 change event
11:03:55.269 -> >>> [ 251368] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:03:55.269 -> >>> [ 251375] HomeKit: [Client 1073696988] Got characteristic 1.10 change event
11:03:56.045 -> !!! [ 252147] HomeKit: [Client 1073696988] socket.write, data_size=167, write_size=94
11:03:56.045 -> >>> [ 252155] HomeKit: [Client 1073696988] Closing client connection
11:03:56.383 -> !!! [ 252464] HomeKit: [Client 0] The socket is null! (or is closed)
11:04:04.766 -> HTTP connected to Drivent 192.168.1.139
11:04:05.340 -> Temp:23.30, Hum:36, Pres:768, CO2:731, SW:1
11:04:05.373 -> Free heap:16032, HomeKit client:1
11:04:05.373 -> HomeKit reporting
11:04:05.373 -> >>> [ 261453] HomeKit: [Client 1073697372] Got characteristic 1.13 change event
11:04:05.373 -> >>> [ 261468] HomeKit: [Client 1073697372] Got characteristic 1.16 change event
11:04:05.373 -> >>> [ 261475] HomeKit: [Client 1073697372] Got characteristic 1.20 change event
11:04:05.373 -> >>> [ 261482] HomeKit: [Client 1073697372] Got characteristic 1.19 change event
11:04:05.373 -> >>> [ 261489] HomeKit: [Client 1073697372] Got characteristic 1.10 change event
11:04:05.946 -> !!! [ 262047] HomeKit: [Client 1073697372] socket.write, data_size=167, write_size=94
11:04:05.946 -> >>> [ 262055] HomeKit: [Client 1073697372] Closing client connection
11:04:06.282 -> !!! [ 262363] HomeKit: [Client 0] The socket is null! (or is closed)
11:04:14.747 -> HTTP connected to Drivent 192.168.1.139
11:04:15.219 -> Temp:23.30, Hum:41, Pres:768, CO2:731, SW:1
11:04:15.219 -> Free heap:16800, HomeKit client:0
11:04:15.254 -> HomeKit reporting
11:04:24.746 -> HTTP connected to Drivent 192.168.1.139
11:04:25.181 -> Temp:23.30, Hum:40, Pres:768, CO2:732, SW:1
11:04:25.181 -> Free heap:16784, HomeKit client:0
11:04:25.181 -> HomeKit reporting
`
How can this be avoided?

AGRO666, 2021-08-25:
I think the problem is that the keepalive parameter is too high. The program continues to send packets to the client when the client is no longer online. And when the program realizes that the client is disconnected, the buffer already contains 5-6 packets, and the buffer is not completely cleared. I was able to avoid reducing the heap by setting the parameters:
#define HOMEKIT_SOCKET_KEEPALIVE_IDLE_SEC      5 // Instead of 180
#define HOMEKIT_SOCKET_KEEPALIVE_INTERVAL_SEC  3 //Instead of 30
#define HOMEKIT_SOCKET_KEEPALIVE_IDLE_COUNT     1 //Instead of 4
but it seems to me that this is the wrong decision

borissov, 2022-07-19:
I have the same issue. Do you find a solution?

AGRO666, 2022-07-19:
Try:
        HOMEKIT_SERVICE(THERMOSTAT, .characteristics = (homekit_characteristic_t*[]) {
            &currentHeatingCoolingState,
            &targetHeatingCoolingState,
            &currentTemperature,
            &targetTemperature,
            &temperatureDisplayUnit,
            NULL
        }),
        HOMEKIT_SERVICE(HUMIDITY_SENSOR, .characteristics=(homekit_characteristic_t*[]) {
            &currentRelativeHumidity,
            NULL
        }),

speak russian?

AGRO666, 2022-07-20:
my code:
 
    if (temp_hk != temp)
    {
      homekit_characteristic_notify(&cha_temperature, cha_temperature.value);
      delay(100);
    }
    if (hum_hk != hum)
    {
      homekit_characteristic_notify(&cha_humidity, cha_humidity.value);
      delay(100);
    }
    temp_hk = temp;
    hum_hk = hum;
      ......
      only 1 send per 60 sec (when change something)

And i use Core 2.7.4

AGRO666, 2022-07-20:
And remove from the code
HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),

borissov, 2022-07-31:
I add checks for differences in characteristics and notify only when is changed. thanks!