Mixiaoxiao/Arduino-HomeKit-ESP8266 #149: Custom Characteristic - compatible with EVE App

Camaendir, 2021-09-29:
Hey, I want to write a simple LED-Strip with Animation support. The EVE App supports custom characteristics. How can i add a custom one for my animation state? I've tried adding a `HOMEKIT_DECLARE_CHARACTERISTIC_ANIMATION` to the `characteristics.h`, but everytime i try to compile that, it gives me an "initializer element is not constant" error. Is there a way to add a custom characteristic? If so does anyone have an example?

Thanks for the help.

sonicpixel2020, 2021-10-20:
Iam also interested in a custom characteristic example. For example a liquid level.

adityapal-singh-chauhan, 2023-06-18:
i also have been trying but to no avail. I want a push button in the app, and so to do it how can i make it happen ?