Mixiaoxiao/Arduino-HomeKit-ESP8266 #154: Cannot find device after removing it

hitenlulla, 2021-11-20:
Hello, I was trying something and I removed the device from my homekit by mistake, Now I'm not able to get that device on the homekit app. One solution I found was to reupload the code but it's not possible to do this again and again. As I'm planning to put this device into my ceiling and I will not be able to remove it. Is there anything that I can add in the code to fix this?

morkiuz, 2021-12-07:
Same here! Even after reflashing esp and changing properties of the device, it still does not appear on the "select accessory to add device" list.

morkiuz, 2021-12-07:
found the answer:

https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/73#issuecomment-744017614

hitenlulla, 2021-12-07:
@morkiuz I think they have stopped maintaining this library, If you find any solution let me know, If I'll find the solution I'll let you know. You can mail me at hitenlulla99@gmail.com

hitenlulla, 2021-12-07:
@morkiuz Have you tried this?

morkiuz, 2021-12-07:
Yeah, works fine. Just add a call to this function in setup(). I would upload it like this and re-upload with this function call commented out, just so if it loses power it doesn't delete everything from memory.

morkiuz, 2021-12-07:
actually abandoned this lib, pairs well with my phone but then goes unresponsive. It sometimes appears on my old iPad with current data, but then goes offline again. Not sure if it's this library or some Homekit BS, I'm rewriting the code for the sensor for it to work with HA.

hitenlulla, 2021-12-17:
To solve that unresponsive issue, use

#include <ESP8266mDNS.h>
bool announce(){
  MDNS.announce();
  return true;
}

And call this announce() function in fixed intervals, I'm calling it in every 10 seconds to announce the connection of the ESP to the WIFI and Homekit Library.

Also, Thank you for homekit_storage_reset();
It solves the problem of reconnecting, I connected a Switch and set a trigger for double click to call this function, This way I don't need to upload the code again.

Would you mind connecting on any social media so we can discuss about this 'HA'? It sounds interesting.

morkiuz, 2021-12-18:
Thanks for the unresponsiveness fix. See PM

closed 2022-01-02 by hitenlulla