Mixiaoxiao/Arduino-HomeKit-ESP8266 #158: Device name + mac address

askarkurymbayev, 2021-12-21:
Kind time of the day friends, I can not solve the problem with the automatic name of devices in the line
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_ (NAME, "name" + MAC address);
can't bind

what i tried to do
     int name_len = snprintf (NULL, 0, "Outlet-% 02X% 02X% 02X",
                             macaddr [3], macaddr [4], macaddr [5]);
     char * name_value = malloc (name_len + 1);
     snprintf (name_value, name_len + 1, "Outlet-% 02X% 02X% 02X",
              macaddr [3], macaddr [4], macaddr [5]);
HOMEKIT_CHARACTERISTIC_ (NAME, name_value);

hitenlulla, 2022-03-07:
Did you solve it?

ruleechen, 2022-03-21:
My solution is to change the value of cha_name during arduino setup.
`
String hostName;

extern "C" homekit_characteristic_t cha_name;

void setup(void) {
  hostName = WiFi.macAddress();
  cha_name.value.string_value = const_cast<char*>(hostName.c_str());
}

void loop(void) { }

`

LouisLee985, 2022-06-23:
> Kind time of the day friends, I can not solve the problem with the automatic name of devices in the line homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_ (NAME, "name" + MAC address); [...]

try this:
```
uint8_t mac[WL_MAC_ADDR_LENGTH];
	WiFi.macAddress(mac);
	int name_len = snprintf(NULL, 0, "%s_%02X%02X%02X", "Outlet", mac[3], mac[4], mac[5]);
	char *name_value = (char*)malloc(name_len + 1);
	snprintf(name_value, name_len + 1, "%s_%02X%02X%02X", "Outlet", mac[3], mac[4], mac[5]);
```