Mixiaoxiao/Arduino-HomeKit-ESP8266 #159: Blinds 28byj-48 uln2003

exrector, 2021-12-29:
Someone help make a sketch.
Please.

AugustinMauroy, 2022-07-26:
The best is to write directly how you can help. Because at the moment there are people who can help you but they don't know how.
Could you be more clear.

Have good day!
Augustin Mauroy

exrector, 2022-08-05:
> Лучше всего написать прямо, как вы можете помочь. Потому что на данный момент есть люди, которые могут тебе помочь, но не знают как. Не могли бы вы быть более ясным.
> 
> Хорошего дня! Огюстен Морой
Hi, @AugustinMauroy 
I found the only working project for blinds.  But this is for homebridge. https://www.studiopieters.nl/homebridge-mqtt-blinds/
I asked someone to stitch these two homekit projects together.  I cant.  Does not work.  I would like to use exactly 28byj-48 and uln2003.