Mixiaoxiao/Arduino-HomeKit-ESP8266 #160: REMOTE_KEY characteristic resets ESP

hitenlulla, 2022-01-01:
Hello, I'm trying to make a Television Remote using this library. I managed to create a toggle button to turn on and turn off the TV. But I want to use the remote accessory (You can find it in control center) to use the arrow keys and media buttons. I figured that I need to enable REMOTE_KEY characteristic in TELEVISION service. But doing that resets the ESP. Following is the code and Serial log.

TV.ino file
`
#include <Arduino.h>
#include <arduino_homekit_server.h>
#include <ESP8266WiFi.h>

#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

void setup() {
	Serial.begin(115200);
	WiFi.begin("SSID","PASS");
	homekit_storage_reset(); // to remove the previous HomeKit pairing storage when you first run this new HomeKit example
	my_homekit_setup();
}

void loop() {
	my_homekit_loop();
}

// access your HomeKit characteristics defined in my_accessory.c
extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_active;
extern "C" homekit_characteristic_t cha_remote_key;

void cha_remote_key_setter(const homekit_value_t value) {
  Serial.println("Remote Key");
  Serial.println(value.float_value);
}

void cha_active_setter(const homekit_value_t value) {
  Serial.println("Active");
  Serial.println(value.bool_value);
}

void my_homekit_setup() {
	cha_remote_key.setter = cha_remote_key_setter;
  cha_active.setter = cha_active_setter;
	arduino_homekit_setup(&config);
}

void my_homekit_loop() {
	arduino_homekit_loop();
}
`

my_accesory.c file:
`
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

homekit_characteristic_t cha_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t cha_active_identifier = HOMEKIT_CHARACTERISTIC_(ACTIVE_IDENTIFIER, "");
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "ESP_TV");
homekit_characteristic_t cha_sleep_discovery_mode = HOMEKIT_CHARACTERISTIC_(SLEEP_DISCOVERY_MODE, 1);
homekit_characteristic_t cha_remote_key = HOMEKIT_CHARACTERISTIC_(REMOTE_KEY, 1);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_television, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "TV"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
		HOMEKIT_SERVICE(TELEVISION, .primary=true, .characteristics=(homekit_characteristic_t*[]){
		  &cha_active,
			&cha_active_identifier,   
			&cha_name,
      &cha_sleep_discovery_mode, 
      &cha_remote_key,
			NULL
		}),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};
`

Serial Monitor Debug:
`
>>> [     74] HomeKit: Using existing accessory ID: 5C:53:48:C5:24:2B
>>> [     80] HomeKit: Found admin pairing with B6C1370D-8D9B-430C-BD24-1A13F3AB6E7F, disabling pair setup
>>> [     90] HomeKit: Configuring MDNS
>>> [     93] HomeKit: Init server over
>>> [   4448] HomeKit: WiFi connected, ip: 192.168.5.26, mask: 255.255.255.0, gw: 192.168.5.1
>>> [   4457] HomeKit: Configuring MDNS
>>> [   4462] HomeKit: MDNS begin: TV, IP: 192.168.5.26
>>> [    375] HomeKit: Formatting HomeKit storage at 0xfb000
>>> [    423] HomeKit: Starting server
>>> [    427] HomeKit: Resetting HomeKit storage
>>> [    431] HomeKit: Formatting HomeKit storage at 0xfb000
>>> [    479] HomeKit: Generated new accessory ID: 20:99:20:C9:0A:EB
>>> [    538] HomeKit: Generated new accessory key
>>> [    543] HomeKit: Preiniting pairing context
>>> [    548] HomeKit: Using user-specified password: 111-11-111
>>> [    568] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [   8196] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  11621] HomeKit: Preinit pairing context success
>>> [  11626] HomeKit: Configuring MDNS
>>> [  11631] HomeKit: MDNS begin: TV, IP: 192.168.5.26
>>> [  11637] HomeKit: Init server over
>>> [  27347] HomeKit: Got new client: local 192.168.5.26:5556, remote 192.168.5.11:55401
>>> [  27357] HomeKit: [Client 1073684524] Pair Setup Step 1/3
>>> [  29775] HomeKit: [Client 1073684524] Pair Setup Step 2/3
>>> [  29786] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  39752] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  45121] HomeKit: [Client 1073684524] Pair Setup Step 3/3
>>> [  45150] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  45871] HomeKit: Added pairing with B6C1370D-8D9B-430C-BD24-1A13F3AB6E7F
>>> [  45944] HomeKit: Free saved_preinit_pairing_context
>>> [  45949] HomeKit: [Client 1073684524] Successfully paired
>>> [  46144] HomeKit: [Client 1073684524] Disconnected!
>>> [  46150] HomeKit: [Client 1073684524] Closing client connection
>>> [  46192] HomeKit: Got new client: local 192.168.5.26:5556, remote 192.168.5.11:55402
>>> [  46202] HomeKit: [Client 1073681876] Pair Verify Step 1/2
>>> [  46522] HomeKit: Free heap: 41856
>>> [  47173] HomeKit: [Client 1073681876] Pair Verify Step 2/2
>>> [  47180] HomeKit: [Client 1073681876] Found pairing with B6C1370D-8D9B-430C-BD24-1A13F3AB6E7F
>>> [  47203] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  47934] HomeKit: [Client 1073681876] Verification successful, secure session established
>>> [  47942] HomeKit: Free heap: 41968
>>> [  48004] HomeKit: [Client 1073681876] Get Accessories

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40217475 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffc0 offset: 0190
3ffffe30:  3fff1241 00000001 3fff1bc4 4021746d  
3ffffe40:  3fff1285 00000001 00000001 00000000  
3ffffe50:  3ffe9307 3ffe860c 3fff1bc4 3ffe8604  
3ffffe60:  3fff0c44 00000000 3fff1bc4 3ffe8604  
3ffffe70:  3fff0c44 3ffe8634 3fff1bc4 3ffe8604  
3ffffe80:  3fff0c44 3ffe8634 3fff1bc4 40219b3b  
3ffffe90:  3ffe85e8 3ffe8910 3ffffef0 00000049  
3ffffea0:  3fff1e54 00000020 3fff1e54 00010000  
3ffffeb0:  3fff1e8a 3fff1044 3fff0c44 4021b441  
3ffffec0:  3fff1e64 00000037 3fff107c 40204c04  
3ffffed0:  3fff1e54 3ffef6a8 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  3fff1e8b 00000001 00000001 3fff1e87  
3fffff00:  3fff1e54 00000030 00000000 3fff1e76  
3fffff10:  00000000 3ffef700 3fff2138 3fff0c54  
3fffff20:  3fffdad0 00000049 00000020 3ffef6e0  
3fffff30:  3fff1e54 3fff1e54 3fff0c44 402182ed  
3fffff40:  00000037 1a05a8c0 3fff2138 3ffef858  
3fffff50:  00000037 000015b4 3ffe85c8 3ffef858  
3fffff60:  3fffdad0 3ffef700 3ffef844 3ffef858  
3fffff70:  3fffdad0 3fff0924 3fff1044 4021a191  
3fffff80:  00000000 00000000 3ffef6dc 4021b4d6  
3fffff90:  3fffdad0 00000000 3ffef844 40216c9c  
3fffffa0:  feefeffe feefeffe feefeffe 40223060  
3fffffb0:  feefeffe feefeffe 3ffe8b84 40100e81  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v000743e0
~ld
`

Output:
[image]
[image]

closed 2022-01-19 by hitenlulla

boboxx, 2023-03-26:
Did you ever find a solution? I think I have the same problem when calling homekit_storage_reset();

22:02:57.009 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:02:57.009 -> 
22:02:57.009 -> Exception (9):
22:02:57.009 -> epc1=0x4000425c epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe9c9b depc=0x00000000
22:02:57.009 -> 
22:02:57.009 -> >>>stack>>>

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4000425c
EXCVADDR: 0x3ffe9c9b

Decoding stack results
0x40205fc1: homekit_storage_set_magic at C:\Users\user\Documents\Arduino\libraries\Arduino-HomeKit-ESP8266\src\storage.c line 105
0x40206022: homekit_storage_init at C:\Users\user\Documents\Arduino\libraries\Arduino-HomeKit-ESP8266\src\storage.c line 124
0x4020603c: homekit_storage_init at C:\Users\user\Documents\Arduino\libraries\Arduino-HomeKit-ESP8266\src\storage.c line 126
0x40227ae3: homekit_server_init(homekit_server_config_t*) at C:\Users\user\Documents\Arduino\libraries\Arduino-HomeKit-ESP8266\src\arduino_homekit_server.cpp line 3304
0x4023693c: snprintf at /workdir/repo/newlib/newlib/libc/stdio/snprintf.c line 90

hitenlulla, 2023-07-14:
@boboxx yes I did, the issue was how I was configuring the library. I can share my code if you want for reference. Just drop your mail ID. I'll send you the .ino file

Djtekken, 2024-02-05:
> @boboxx yes I did, the issue was how I was configuring the library. I can share my code if you want for reference. Just drop your mail ID. I'll send you the .ino file

Can u share the file please? thanks u so much. Djtekken2005@gmail.com