Mixiaoxiao/Arduino-HomeKit-ESP8266 #162: PWMRANGE error

DziveQ, 2022-02-02:
Hi. I'm trying to use example simple_led and the only thing I changed was SSID name and password. When I'm uploading, I'm getting this error:

C:\Users\kacpe\AppData\Local\Temp\arduino_modified_sketch_215244\simple_led_accessory.c: In function 'led_update':
simple_led_accessory.c:67:13: error: 'PWMRANGE' undeclared (first use in this function)
   67 |   int pwm = PWMRANGE - (int) (led_bri * 1.0 * PWMRANGE / 100.0 + 0.5f);
      |             ^~~~~~~~
C:\Users\kacpe\AppData\Local\Temp\arduino_modified_sketch_215244\simple_led_accessory.c:67:13: note: each undeclared identifier is reported only once for each function it appears in
exit status 1
'PWMRANGE' undeclared (first use in this function)

I'm using correct Port and Board settings (it's working with other sketches)

AugustinMauroy, 2022-05-14:
I have same problem

kuznetsov-m, 2022-05-17:
Guys, I have the same problem.
Googling for the PWMRANGE variable gave me the following [issue](https://github.com/esp8266/Arduino/issues/5957):
> 1.) the change of PWMRANGE via analogWriteRange(new_range) is not possible!
> -> if you set 1000 or 255 for example there ist no change -> only 1023.

This issue has nothing to do with this project. However, I decided to try redefining the PWMRANGE variable. 
I just added a declaration `int PWMRANGE = 1023;` to the method [led_update()](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/blob/master/examples/legacy/simple_led/simple_led_accessory.c#L65-L74) 
```
void led_update() {
	if (led_power) {
		int PWMRANGE = 1023;     // redefining PWMRANGE here
		int pwm = PWMRANGE - (int) (led_bri * 1.0 * PWMRANGE / 100.0 + 0.5f);
		analogWrite(PIN_LED, pwm);
		printf("ON  %3d (pwm: %4d of %d)\n", led_bri, pwm, PWMRANGE);
	} else {
		printf("OFF\n");
		digitalWrite(PIN_LED, HIGH);
	}
}
```

This is a very dirty solution. But it allowed me to compile the example.

dani34dag, 2022-09-09:
Con esta corrección me ha funcionado, el problema es que al poner el 100% de brillo se apaga, y al poner 0% se enciende del todo, alquien sabe algo? en definitiva funciona del revés

FrenchBen, 2022-09-13:
This is due to an old reference, for the esp8266 library that was since updated.
`#define PWMRANGE 1023` was removed
https://github.com/esp8266/Arduino/pull/7456/

In the default library this was set to the upstream value of 255:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring_pwm.cpp#L29