Mixiaoxiao/Arduino-HomeKit-ESP8266 #165: Switch always reboot as OFF

JeroenV1982, 2022-02-21:
Sorry I'm not a code expert but by looking at the code, shouldnt it return to its last known state after a power loss?

If not, how to modify the code to make it reboot as SWITCH ON after a reboot?

hitenlulla, 2022-03-07:
When your device restarts it will always go back to the off state, you can store the values over the internet (I use MQTT) or in SPIFFS (Internal file system) so that your device takes the stored value and not the default off value when it restarts.

JeroenV1982, 2022-03-09:
> When your device restarts it will always go back to the off state, you can store the values over the internet (I use MQTT) or in SPIFFS (Internal file system) so that your device takes the stored value and not the default off value when it restarts.
It would be great if this feature could be integrated in the master. For now, do you know how I can hard-code it to start as ON?

hitenlulla, 2022-03-09:
https://youtu.be/pfJROpQg-Is
This should be a good start point to get used to SPIFFS and Arduino JSON.
Try it out and still if you don't find a solution, connect with me at https://www.linkedin.com/in/hiten-lulla-a434201aa