Mixiaoxiao/Arduino-HomeKit-ESP8266 #166: MDNS connection failed after router restarts.

hitenlulla, 2022-03-13:
Hello,
I found this small problem, When the ESP8266 is on and the router goes off, on reconnection with router the Homekit accessories become 'not responding'. And when I restart the ESP, the accessories are back online.
ps, **I have used MDNS.announce() for every 10 seconds and everything works properly except this edge case.**

I found a temporary solution though, it looks like when the connection is found after being disconnected, the MDNS service starts but the connection to the client is not made, I removed the line 3169 in arduino_homekit_server.cpp which is a return statement, to make the function always start making connection to the client when the device gets reconnected.

```
// arduino_homekit_server.cpp : line 3164
if (homekit_mdns_started) {
		MDNS.close();
		MDNS.begin(name->value.string_value, staIP);
		INFO("MDNS restart: %s, IP: %s", name->value.string_value, staIP.toString().c_str());
		MDNS.announce();
		// return;                                                      **<- Remove this line**
}
```

But this leads to blocking of Loop when the internet in the router is lost and I cannot afford that. Hence this workaround was a big flop. Can anyone help me solve this problem?

ruleechen, 2022-03-21:
I encountered this issue before. My solution is to comment out this line.
`// MDNS.close();`

hitenlulla, 2022-03-21:
> I encountered this issue before. My solution is to comment out this line.
> 
> `// MDNS.close();`

Is this solution stable? Have you tested it in the long run?

ruleechen, 2022-03-21:
[image]

ruleechen, 2022-03-21:
https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/139

hitenlulla, 2022-03-21:
> https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/139

Hello sir, do you mind connecting on any social media? I'm new into this technology and I want to learn more about it.

ruleechen, 2022-03-22:
> > #139
> 
> Hello sir, do you mind connecting on any social media? I'm new into this technology and I want to learn more about it.

Sorry I don't have a global social media. Github would be fine for us.

ivishaltejwani, 2022-03-24:
> ![0DA81267-845D-4C72-AAB0-7D5287519386](https://user-images.githubusercontent.com/907369/159244079-c88d8731-373e-43f0-801d-5078b389a459.png

can you tell whats the app youre using in the screenshot have been struggling with ota

ivishaltejwani, 2022-03-24:
> [image]

hey would you be able to share which app is this, have been struggling with ota

ruleechen, 2022-03-25:
> hey would you be able to share which app is this, have been struggling with ota

Please refer to this repo. It is a framework build base upon esp8266. It includes some common used functionalities as you can see them in the top menu of the screenshot.
https://github.com/ruleechen/home-esp8266