Mixiaoxiao/Arduino-HomeKit-ESP8266 #167: Multiple motion sensors on one 8266

micahvdm, 2022-03-18:
Is it possible to have multiple motion sensors on one device and assign then to different rooms? I have looked at the multi accessory example, but I'm not sure how I would go about having a few (say 5 sensors) on the same device.

Any help would be greatly appreciated!

ruleechen, 2022-03-21:
Yes, I think it can be achieved. As you can see in the multi accessory example we can define some sensor accessories and export them via a bridge accessory. That is what this library done. On the other side, Apple Home app add the bridge accessory then bring in the sensor accessories and then we can assign them  to different rooms.

In a nutshell, assign accessories to different rooms is the ability of Apple Home app.

PimV5755, 2022-04-26:
Hi, I have cracked that one of multiple accessories. I can post an example.
It is not that hard once you understand the indexing.

AugustinMauroy, 2022-07-26:
Hello
Would you add your code (to the repository thanks to a pull request) ? to increase the number of examples.

PimV5755, 2022-07-27:
Sure. I have many more. I’ll group them in a batch.