Mixiaoxiao/Arduino-HomeKit-ESP8266 #168: Light Sensor pairing fails at Get Accessories

DanMarkwood, 2022-03-29:
I first tried the Switch example and that worked great. I then attempted to write a switch for a Light Sensor. When adding the sensor to the Home app, it fails at Get Accessories, log attached.

This is the contents of my-accessories.c. I am hoping you can help me find my error.

/*
 * my_accessory.c
 * Define the accessory in C language using the Macro in characteristics.h
 *
 *  Created on: 2020-05-16
 *      Author: Mixiaoxiao (Wang Bin)
 */

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

// format: float; min 0.0001, max 100000
homekit_characteristic_t cha_light = HOMEKIT_CHARACTERISTIC_(CURRENT_AMBIENT_LIGHT_LEVEL, 1);
// format: string; HAP section 9.62; max length 64
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "Light Sensor");

homekit_accessory_t *accessories[] = {
   HOMEKIT_ACCESSORY(.id=4, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {
  HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Light Sensor"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),    	
HOMEKIT_SERVICE(LIGHT_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
			&cha_name,
			&cha_light,
			NULL
		}),
		NULL
	}),
	
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};

[Log.txt](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/8374151/Log.txt)

DanMarkwood, 2022-03-30:
I got this to work.

tickietackie, 2022-06-07:
How did you solve it?