Mixiaoxiao/Arduino-HomeKit-ESP8266 #174: Heater_Cooler Service: temperature controller wheel not showing up in app

gutasiadam, 2022-06-05:
I am building an AC Controller NodeMCU, that will control an AC Through IR.

I wrote the following code for it:

```
homekit_characteristic_t cha_switch_on2 = HOMEKIT_CHARACTERISTIC_(ON, false);
homekit_characteristic_t cha_ac_active = HOMEKIT_CHARACTERISTIC_(ACTIVE,0);
homekit_characteristic_t cha_ac_curr_tmp = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE,20);
homekit_characteristic_t cha_ac_hcCurrentState = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATER_COOLER_STATE,0); //off, heat, cool, auto
homekit_characteristic_t cha_ac_hcTargetState = HOMEKIT_CHARACTERISTIC_(TARGET_HEATER_COOLER_STATE,0);
homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_air_conditioner, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "AC Controller"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "----------"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "00000001"),
            HOMEKIT_CHARACTERISTIC(MODEL, "AC IR"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
    HOMEKIT_SERVICE(HEATER_COOLER, .primary=true, .characteristics=(homekit_characteristic_t*[]){
      &cha_switch_on2,
      &cha_ac_curr_tmp,
      &cha_ac_hcCurrentState,
      &cha_ac_hcTargetState,
      NULL
    }),    
        NULL
    }),
    NULL
};

```
However, as the issue's title indicates, once I upload this code to the controller and set up the home up, I only see the following:
[image]

AugustinMauroy, 2022-07-26:
Do you still have the problem today?

Is it possible to get your code ? In order to find the potential error.

gutasiadam, 2023-01-04:
Yes, my problem still exists. Here are my source code files pasted on _Paste ofCode_
**Example02_Switch.ino**: [paste.ofcode](https://paste.ofcode.org/YwyqTwCgkYPR8XFRgMsJPw)
**my_accessory.c**: [paste.ofcode](https://paste.ofcode.org/gq99TeeaDL9SM5WEtxbmqu)
**wifi_info.h**: [paste.ofcode](https://paste.ofcode.org/JdHgYE2grKDy7MB9KhRxmf)

Borx2001, 2023-10-11:
Hey!
Do you know how to solve it? Im having the same issue. 
Could you please share the code? I can't access the Paste ofCode :(