Mixiaoxiao/Arduino-HomeKit-ESP8266 #181: device no response, serial monitor find 0 clients

loujiabin, 2022-08-12:
esp8266 works perfect once it connected in the beginning, but after several days, it will lost connections. just like those son-off devices i bought from TaoBao.
try to change modem settings such as , use certain channel, 2.4g only... the good news is: son-off devices never lose connections after i configure modem. but esp8266 still no response after a few weeks. 
on esp8266 serial monitor, it saids: homekit clients:0

any suggestion what command i should use in serial monitor to figure our the problem?

thiti-y, 2022-09-01:
I would like to try something. Can you try turn off modem and then turn on again please. I think it not work after that.

So, I think it is root cause of the issue.

thiti-y, 2022-09-11:
@L2Miko I create PR (https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/pull/184) for fix that issue.
Any way, You can try edit code follow PR in your local, I think is solve the issue. that it work for me.

loujiabin, 2022-09-12:
Problem solved.
Found from other comments.
Erase mDNS restart from lib file then no more connection lose

closed 2022-09-12 by loujiabin

blahBlahhhJ, 2022-11-02:
Sorry. I can't seem to find any code related to "mDNS restart" from the library. Do you mean this code block?
```
if (homekit_mdns_started) {
  MDNS.close();
  MDNS.begin(name->value.string_value, staIP);
  INFO("MDNS restart: %s, IP: %s", name->value.string_value, staIP.toString().c_str());
  MDNS.announce();
  return;
}
```
and should the entire block be deleted?

loujiabin, 2022-11-03:
No,  only erase this line

> MDNS.close();
It’s under library file which i cant remember the name.

blahBlahhhJ, 2022-11-05:
thanks! works like a charm, at least for now.