Mixiaoxiao/Arduino-HomeKit-ESP8266 #200: Pairing issue

laszloferenczi, 2022-12-26:
H,

I would like to test Example05 WS2812 Neopixel examples. But after then configured wifi and try to pair device I get the following result.

WiFi connected, IP: 192.168.1.18
>>> [   6003] HomeKit: Formatting HomeKit storage at 0x3fb000
>>> [   7036] HomeKit: Update the CPU to run at 160MHz
>>> [   7042] HomeKit: Starting server
>>> [   7045] HomeKit: Resetting HomeKit storage
>>> [   7049] HomeKit: Formatting HomeKit storage at 0x3fb000
>>> [   7079] HomeKit: Generated new accessory ID: 78:00:5B:CC:F7:14
>>> [   7136] HomeKit: Generated new accessory key
>>> [   7141] HomeKit: Preiniting pairing context
>>> [   7146] HomeKit: Using user-specified password: 111-11-111
>>> [   7167] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  13845] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  17265] HomeKit: Preinit pairing context success
>>> [  17270] HomeKit: Configuring MDNS
>>> [  17275] HomeKit: MDNS begin: ESP8266 Lamp, IP: 192.168.1.18
>>> [  17282] HomeKit: Init server over
Free heap: 40760, HomeKit clients: 0
Free heap: 41312, HomeKit clients: 0
Free heap: 40936, HomeKit clients: 0
>>> [  31336] HomeKit: Got new client: local 192.168.1.18:5556, remote 192.168.1.12:50049
>>> [  31357] HomeKit: [Client 1073683540] Pair Setup Step 1/3
>>> [  31517] HomeKit: [Client 1073683540] Pair Setup Step 2/3
>>> [  31532] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  51544] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Free heap: 38888, HomeKit clients: 1
>>> [  61629] HomeKit: [Client 1073683540] Disconnected!
>>> [  61634] HomeKit: [Client 1073683540] Closing client connection
>>> [  61640] HomeKit: Free saved_preinit_pairing_context
>>> [  61645] HomeKit: [Client 0] Clear the pairing context
>>> [  61661] HomeKit: Preiniting pairing context
>>> [  61665] HomeKit: Using user-specified password: 111-11-111
>>> [  61698] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  75035] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  81838] HomeKit: Preinit pairing context success
Free heap: 41952, HomeKit clients: 0
>>> [  83972] HomeKit: WiFi connected, ip: 192.168.1.18, mask: 255.255.255.0, gw: 192.168.1.254
>>> [  83981] HomeKit: Configuring MDNS
>>> [  83988] HomeKit: MDNS restart: ESP8266 Lamp, IP: 192.168.1.18

And a pairing unsuccess.
Please help!
Laci

gvuyk, 2023-02-04:
Hi,
If you are not on the latest version, try it, [v1.4.0](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/releases/tag/v1.4.0) has something that could help with this.
If that is not enough, try disabling or re-scheduling any extra function that could be using CPU during the pairing process

rayenking, 2023-08-17:
hey I have the same issue when pairing, it say the setup code is incorrect in the app