Mixiaoxiao/Arduino-HomeKit-ESP8266 #205: homekit_accessory_category_faucet

gielk, 2023-01-27:
Is there anyone on GitHub who has an example or can show the code for building an ESP8266 HomeKit faucet? I was able to make a switch, but I am having trouble understanding how to convert it to a faucet. I did see the 'homekit_accessory_category_faucet' category, but I am not sure how to proceed.

angel41671, 2023-04-05:
You should use the "ACTIVE" characteristic (uint8) instead of "ON" (bool). I improvised a configuration code for a faucet (didn't try it), it may be useful:

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

// FAUCET
// required: ACTIVE
// optional: NAME, STATUS_FAULT

// format: uint8; 
homekit_characteristic_t cha_faucet_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_faucet, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, “Faucet”),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
		HOMEKIT_SERVICE(FAUCET, .primary=true, .characteristics=(homekit_characteristic_t*[]){
			&cha_faucet_active,
			NULL
		}),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};

gielk, 2023-04-05:
Awesome I am going to try it when I have time.

pjmeijer, 2023-06-23:
Hey, thanks for inspiring me. I wanted to do something similar and I got this to work with some tinkering:

`
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

// Switch (HAP section 8.38)
// required: ACTIVE
// optional: NAME

// format: bool; HAP section 9.70; write the .setter function to get the switch-event sent from iOS Home APP.
homekit_characteristic_t cha_faucet_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);
homekit_characteristic_t cha_faucet_active2 = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);
homekit_characteristic_t cha_faucet_program_mode = HOMEKIT_CHARACTERISTIC_(PROGRAM_MODE, false);
homekit_characteristic_t cha_faucet_program_mode2 = HOMEKIT_CHARACTERISTIC_(PROGRAM_MODE, false);
homekit_characteristic_t cha_faucet_in_use = HOMEKIT_CHARACTERISTIC_(IN_USE, false);
homekit_characteristic_t cha_faucet_in_use2 = HOMEKIT_CHARACTERISTIC_(IN_USE, false);
//homekit_characteristic_t cha_faucet_active3 = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);
//homekit_characteristic_t cha_faucet_active4 = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);

// format: string; HAP section 9.62; max length 64
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "Faucet");

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_faucet, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Havevanding"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
		HOMEKIT_SERVICE(FAUCET, .primary=true, .characteristics=(homekit_characteristic_t*[]){
			&cha_faucet_active,
      &cha_faucet_program_mode,
      &cha_faucet_in_use,
			NULL
		}),
        NULL
    }),
        HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_faucet, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Haveslangen"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
    HOMEKIT_SERVICE(FAUCET, .primary=true, .characteristics=(homekit_characteristic_t*[]){
			&cha_faucet_active2,
      &cha_faucet_program_mode2,
      &cha_faucet_in_use2,
      NULL
    }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
[... 149 lines cut ...]
    }

    // Check if the active state has changed
    if (is_switch_on != cha_faucet_active.value.bool_value) {
      cha_faucet_active.value.bool_value = is_switch_on;
      LOG_D("Faucet changed to: %s", is_switch_on ? "ON" : "OFF");
      homekit_characteristic_notify(&cha_faucet_active, cha_faucet_active.value);
    }
    
    // Check if the active2 state has changed
    if (is_switch2_on != cha_faucet_active2.value.bool_value) {
      cha_faucet_active2.value.bool_value = is_switch2_on;
      LOG_D("Faucet2 changed to: %s", is_switch2_on ? "ON" : "OFF");
      homekit_characteristic_notify(&cha_faucet_active2, cha_faucet_active2.value);
    }

    LOG_D("Free heap: %d, HomeKit clients: %d",
				ESP.getFreeHeap(), arduino_homekit_connected_clients_count());
    LOG_D("PIN_SWITCH %d is in use: %d" ,PIN_SWITCH, cha_faucet_in_use.value.bool_value);
    LOG_D("PIN_SWITCH %d is in use: %d" ,PIN_SWITCH2, cha_faucet_in_use2.value.bool_value);

  }

}
`

pjmeijer, 2023-06-23:
However, looking at this I'm not sure I'm doing it correctly:
https://github.com/apple/HomeKitADK/blob/master/HAP/HAPServiceTypes.h
Here they are talking about valves associated with with another type.

Looking in Arduino-HomeKit_ESP8266 I find this and that's what I used:
#define HOMEKIT_SERVICE_FAUCET HOMEKIT_APPLE_UUID2("D7")

/**
 Defines that the accessory supports the control of a irrigation system.
 
 Required Characteristics:
 - ACTIVE
 - PROGRAM_MODE
 - IN_USE
 
 Optional Characteristics:
 - NAME
 - REMAINING_DURATION
 - STATUS_FAULT
 */

PimV5755, 2023-07-07:
Hi, It is a whole range of irrigation stuff.
This works as the simplest of faucet. I have a 220V valve from Ali that does the work.
[Archive.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/11979957/Archive.zip)

Good luck