Mixiaoxiao/Arduino-HomeKit-ESP8266 #209: HomeKit for Sensibo Air - It is possible ?

mateusmsantin, 2023-02-13:
Hi Guys, there is possible to make a control with ESP8266 ?
I don’t find a exemple to beging tests.
I would like to do something like this
https://support.sensibo.com/l/en/article/yt2lig7u5z-home-kit-for-sensibo-air

I would appreciate any kind of help. Thx.

densh1k, 2023-02-19:
[image]

Could it be this?

mateusmsantin, 2023-02-19:
Sounds good, I will try to do it. 
Thanks...

mateusmsantin, 2023-02-23:
my sketch.ini

`#include <Arduino.h>
#include <arduino_homekit_server.h>
#include "wifi_info.h"
#include <IRremoteESP8266.h>
#include <IRsend.h>

//#define LOG_D(fmt, ...) printf_P(PSTR(fmt "\n") , ##VA_ARGS);
#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

//temperature sensor output pin
int temp_outputpin = A0;

// IR settings
const uint16_t kIrLed = 14; // ESP8266 GPIO pin to use. Recommended: 4 (D2).

#include <ir_Hitachi.h>
IRHitachiAc296 ac(kIrLed); // Set the GPIO to be used to sending the message
#define MODE_AUTO kHitachiAc296Auto
#define MODE_COOL kHitachiAc296Cool
#define MODE_DRY kHitachiAc296Dehumidify
#define MODE_HEAT kHitachiAc296Heat
#define MODE_FAN kHitachiAc296Dehumidify;

#define FAN_AUTO kHitachiAc296FanAuto
#define FAN_MIN kHitachiAc296FanLow
#define FAN_MED kHitachiAc296FanMedium
#define FAN_HI kHitachiAc296FanHigh

// Globals
bool queueCommand = false;
void flipQueueCommand(bool newState) {
Serial.write("Flipping queueCommand to %d\n", newState);
queueCommand = newState;
}

void setup() {
Serial.begin(115200);
initWiFi1(); // in wifi_info.h

homekit_storage_reset(); // to remove the previous HomeKit pairing storage when you first run this new HomeKit example

my_homekit_setup();

//ac_rotation_speed.value.int_value = 100;
Serial.write("HomeKit setup complete. About to start ac.begin()\n");
ac.begin();
}

void loop() {
//drd.loop();
my_homekit_loop();
delay(10);

if (queueCommand)
{
Serial.write("Sending AC Command....\n");
ac.send();
flipQueueCommand(false);
}
}

// defined in my_accessory.c
extern "C" homekit_server_config_t config;

extern "C" homekit_characteristic_t ac_display_units;
extern "C" homekit_characteristic_t ac_current_temp;
extern "C" homekit_characteristic_t ac_target_temp;
extern "C" homekit_characteristic_t ac_current_state;
[... 131 lines cut ...]
      HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
      HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
      NULL
    }),
    HOMEKIT_SERVICE(THERMOSTAT, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
      &ac_current_state,
      &ac_target_state,
      &ac_current_temp,
      &ac_target_temp,
      &ac_display_units,
      NULL
    }),
    NULL
  }),
  NULL
};

homekit_server_config_t config = {
  .accessories = accessories,
  .password = "111-11-111"
};
`
I set to 23 degree to test, but I belive that have some problem at get and set, but I can’t find.

densh1k, 2023-02-23:
I don't quite understand the problem, can you describe it in more detail.

mateusmsantin, 2023-02-23:
The serial log show me:

`>>> [  96282] HomeKit: [Client 1073683732] Successfully paired
>>> [  96450] HomeKit: [Client 1073683732] Disconnected!
>>> [  96455] HomeKit: [Client 1073683732] Closing client connection
>>> [  96472] HomeKit: Got new client: local 192.168.15.13:5556, remote 192.168.15.17:52156
>>> [  96493] HomeKit: [Client 1073678604] Pair Verify Step 1/2
>>> [  96987] HomeKit: Free heap: 41136
>>> [  97023] HomeKit: [Client 1073678604] Pair Verify Step 2/2
>>> [  97030] HomeKit: [Client 1073678604] Found pairing with 8EAB837F-EF08-4B55-ADA0-910E50B341B7
>>> [  97060] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  98503] HomeKit: [Client 1073678604] Verification successful, secure session established
>>> [  98511] HomeKit: Free heap: 41248
>>> [  98567] HomeKit: [Client 1073678604] Get Accessories
CURRENT STATE GETTER CALLED! RESPONDING WITH: 0
TARGET STATE GETTER CALLED! RESPONDING WITH: 0
TARGET TEMP GETTER CALLED! RESPONDING WITH: 0.000000
>>> [  98836] HomeKit: [Client 1073678604] Disconnected!
>>> [  98841] HomeKit: [Client 1073678604] Closing client connection
Current temperature: 23.0`

--------------------------8<------------------------------------------

[image]

[image]

[image]

densh1k, 2023-02-23:
This is a problem due to memory. Make sure you are using the correct Homekit library. Check the recommended IDE settings.
[image]

mateusmsantin, 2023-02-23:
I had configure IDE like your remomendadion, there was some configuration diferente before. 

But the error is the same:

`>>> [  43163] HomeKit: [Client 1073684196] Pair Verify Step 1/2
>>> [  43424] HomeKit: Free heap: 40944
>>> [  43511] HomeKit: [Client 1073684196] Pair Verify Step 2/2
>>> [  43517] HomeKit: [Client 1073684196] Found pairing with 8EAB837F-EF08-4B55-ADA0-910E50B341B7
>>> [  43539] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  44267] HomeKit: [Client 1073684196] Verification successful, secure session established
>>> [  44275] HomeKit: Free heap: 41056
>>> [  44381] HomeKit: [Client 1073684196] Get Accessories
CURRENT STATE GETTER CALLED! RESPONDING WITH: 0
TARGET STATE GETTER CALLED! RESPONDING WITH: 0
TARGET TEMP GETTER CALLED! RESPONDING WITH: 0.000000
>>> [  44584] HomeKit: [Client 1073684196] Disconnected!
>>> [  44590] HomeKit: [Client 1073684196] Closing client connection
Current temperature: 23.0`

The getter and setter I thing is out of range, or wrong.
I read that currente temperatura is out of range, the pairing don't works. I fixed for teste to 23.0, but I don't know how I set the other parametres. 

homekit_characteristic_t ac_current_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATING_COOLING_STATE, 0);
homekit_characteristic_t ac_target_state = HOMEKIT_CHARACTERISTIC_(TARGET_HEATING_COOLING_STATE, 0);
homekit_characteristic_t ac_target_temp = HOMEKIT_CHARACTERISTIC_(TARGET_TEMPERATURE, 0);
homekit_characteristic_t ac_display_units = HOMEKIT_CHARACTERISTIC_(TEMPERATURE_DISPLAY_UNITS, 0);

CURRENT STATE GETTER CALLED! RESPONDING WITH: **0**
TARGET STATE GETTER CALLED! RESPONDING WITH: **0**
TARGET TEMP GETTER CALLED! RESPONDING WITH: **0.000000**

densh1k, 2023-02-23:
**Limits:**

> CURRENT_HEATING_COOLING_STATE: 0, 1, 2
> TARGET_HEATING_COOLING_STATE: 0, 1, 2, 3
> TARGET_TEMPERATURE: 10...38 
> TEMPERATURE_DISPLAY_UNITS: 0, 1

**Try this with your own parameters:**
`void my_homekit_setup() {`
`      ac_current_state.value.float_value = 1;`
`}`

mateusmsantin, 2023-02-23:
Hi densh1k, you found the problem. I changed my code to:

`//ac_active.setter = active_setter;
//ac_current_state.getter = current_state_getter;
//ac_current_state.setter = current_state_setter;

ac_current_state.value.float_value = 1;

//ac_target_state.setter = target_state_setter;
//ac_target_state.getter = target_state_getter;
ac_target_state.value.float_value = 30;

//ac_target_temp.setter = target_temp_setter;
//ac_target_temp.getter = target_temp_getter;
ac_target_temp.value.float_value = 19;`

I fixed, and now, I can pairing the device. 
[image]

Thx!