Mixiaoxiao/Arduino-HomeKit-ESP8266 #210: Humidifier and dehumidifier

densh1k, 2023-02-19:
I use ESP8266 to control the humidifier. How to remove the dehumidifier from homekit?

edWin-m, 2023-04-11:
@densh1k you need to modify the values for `TARGET_HUMIDIFIER_DEHUMIDIFIER_STATE ` and  `CURRENT_HUMIDIFIER_DEHUMIDIFIER_STATE`.  You can try the configuration below.

```c
homekit_characteristic_t hk_tar_humidifier_state = HOMEKIT_CHARACTERISTIC_(TARGET_HUMIDIFIER_DEHUMIDIFIER_STATE, 1,
        .valid_values = {.count=1, .values=(uint8_t[]) {1}});  // 0, 1, 2, 3
homekit_characteristic_t hk_cur_humidifier_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HUMIDIFIER_DEHUMIDIFIER_STATE, 1,
        .valid_values={.count=2, .values = (uint8_t[]) {0, 1}}); // 0, 1, 2
```

densh1k, 2023-04-12:
Thanks, ok