Mixiaoxiao/Arduino-HomeKit-ESP8266 #211: After pairing the device disconects

miguelpereiraesousa, 2023-02-24:
I can setup the device but after that the device does not respond and homekit clients is 0
[image]

jeckorf, 2023-04-21:
I also encountered this problem

DayOut, 2023-04-26:
Same problem. First it was fixed by pull request here (just used this repo instead of Mixiaoxiao) https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/pull/212 but today i got this problem again and searching the reason and how to fix it

bigfou, 2023-08-15:
我也是这样的问题，搞了好久，还没找到原因。

thncode, 2023-08-16:
I do have disconnects regularly as well, but automated restarts based on homekit_clients = 0, this works since many months quite well...

ciananDR, 2023-11-15:
I am experiencing the same problem, has anybody found a solution?