Mixiaoxiao/Arduino-HomeKit-ESP8266 #214: Error when I try to Verify/Compile package for ESP8266

Z-Beeblebrox1978, 2023-03-09:
I  trying to compile package for Apple HomeKit for IKEA VINDRIKTNING PM2.5 Air Quality sensor project. And receiving error which pointing to the following package "Arduino HomeKit ESP8266" here is error message(s);
1. /Users/poweruser/Documents/Arduino/libraries/HomeKit-ESP8266/src/arduino_homekit_server.h:5:10: **fatal error: WiFiServer.h:** No such file or directory
2. /Users/poweruser/Documents/Arduino/libraries/HomeKit-ESP8266/src/WiFiClient.h:24:10: fatal error: memory: No such file or directory
 #include <memory>
Sketch library installed:
a) ArduinoOTA
b) WiFiManager
c) Arduino HomeKit ESP8266
d) Apple HomeKit for IKEA VINDRIKTNING PM2.5 Air Quality sensor

Any suggestions will be greatly appreciated.

N5NMR, 2023-04-21:
Hey. 
For Issue 1: Did you install the WiFiServer Bibliothek? 

J.

Z-Beeblebrox1978, 2023-07-24:
all libs has been installed. ***@***.*** and ***@***.***
[image: Screenshot 2023-07-24 at 06.43.12.png]