Mixiaoxiao/Arduino-HomeKit-ESP8266 #218: Television service?

vmarks, 2023-03-31:
Hi, 

Does this support the television service?

I would like to set inputs, volume up/down/mute and power on/off for a television with this project.

hassbian-ABC, 2023-05-19:
yes

PimV5755, 2023-07-07:
Pay attention that you cannot control this accessory from the Home app or using Siri.
Other apps wil work. Apple does not allow uncertified (OpenSource HAP) accessories to audio and video. (Shame)

vmarks, 2023-07-07:
That is incorrect.

Apple allows ravensystems/HAA ir blasters to control television input and
volume for example in home and the control
Center remote control widget.

PimV5755, 2023-07-07:
That would be great. I tried to make a "speaker" accessory to control the volume of an amplifier/speaker. The accessory paired fine but in the Home-app there came the message that it is not supported. Do you have made it working?

Here ther is also the same problem mentioned (few years ago) 
https://github.com/homebridge/homebridge/issues/1326

vmarks, 2023-07-07:
> homebridge/homebridge#1326

I have an IR blaster running ravensystems/haa that controls inputs and
volume of a samsung TV.

Separately, I have homebridge running an onkyo receiver plugin which
appears to homekit like a TV for inputs, and uses a fan speed or lamp
dimmer (config option) to add a volume slider to homekit. Without that, you
have to open control center, tap on remote, and use the volume buttons to
control the volume of the receiver or TV.

but… it works.

PimV5755, 2023-07-07:
Yes, through Homebridge it works. However, I prefer the native setup so
that I can use Siri. BTW I made a HomeKit “gateway” to control my Denon AVR

vmarks, 2023-07-07:
I agree, I prefer to not use homebridge. But the point I was making was
that open source HAP works with volume.

I would like to replicate your Denon control. Are you able to share how you
did it? I was thinking of using ravensystem/haa and mapping the TV
configuration to the AVR urls that Denon uses.

PimV5755, 2023-07-07:
Sure. To make it work with Siri I made it a Lightbulb and limited the
control URL to on/off and volume. I did make the double direction work. So
if you use the remote or the volume knob, and the on/off it does update
HomeKit (as it should)
From that you can add more if you go for your solution. I will post an
archive with th sample when I am home.

hassbian-ABC, 2023-07-07:
I used IRremoteESP8266 and created a TV service

hassbian-ABC, 2023-07-07:
[image]

hassbian-ABC, 2023-07-07:
[image]

hassbian-ABC, 2023-07-07:
[image]

hassbian-ABC, 2023-07-07:
[image]

PimV5755, 2023-07-07:
Nice.

bartosz-777, 2024-03-01:
@hassbian-ABC could you share the code for tv remote(im interested in HomeKit part)? im trying to make it and if you already done it it'd be easier for me