Mixiaoxiao/Arduino-HomeKit-ESP8266 #224: homekit_accessory_category_faucet

vikingY, 2023-05-14:
hi,

I used the flowing code in order to get faucet status (open / close)
from some reason in HomeKit its appeared as updating
please advise :-)
Thanks in advanced. 

**my.accessory.c**

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

homekit_characteristic_t cha_faucet = HOMEKIT_CHARACTERISTIC_(ACTIVE, false);

homekit_accessory_t *accessories[] = {
 // Faucet 29
	HOMEKIT_ACCESSORY(.id=29, .category=homekit_accessory_category_faucet, .services=(homekit_service_t*[]) { 
    	HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
			HOMEKIT_CHARACTERISTIC(NAME, "Water Flow Sensor"),
      HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HomeKit"),
      HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
      HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
			HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
			NULL
		}),
    HOMEKIT_SERVICE(FAUCET, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
			//HOMEKIT_CHARACTERISTIC(NAME, "Faucet"),
			&cha_faucet,
			NULL
		}),
		NULL
	}),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};

Faucet.ino

extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_faucet;

void my_homekit_report() {

    cha_faucet.value.uint8_value = sensorVal; // read from Input set as int
    homekit_characteristic_notify(&cha_faucet, cha_faucet.value);
}

AGRO666, 2023-05-17:
sensorVal = 0 (inactive) or 1 (active). Other values cannot be used
and replace
homekit_characteristic_t cha_faucet = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);

vikingY, 2023-05-18:
hi,
that for your answering and help, unfortuntly I'm still getting 
No Response upon peering to HomeKit.

attch the code, what i'm doing wrong?

[WaterSensor.zip](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/11509155/WaterSensor.zip)

THX :-)
Y@ir