Mixiaoxiao/Arduino-HomeKit-ESP8266 #226: Issue with EEPROM definitions?

Paul2357, 2023-05-29:
So I'm using this to make a DIY homekit blind system and after debugging my code and fixing errors, I'm now left with this error message:

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-build-3009112EB8599B216613EB82652E81F5/libraries/HomeKit-ESP8266/storage.c.o:(.text.find_empty_block+0x0): undefined reference to `_EEPROM_start'

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino-build-3009112EB8599B216613EB82652E81F5/libraries/HomeKit-ESP8266/storage.c.o: in function `homekit_storage_reset':

/mnt/create-efs/webide/29/f3/29f32a690ec98f2e231ae59981834f97:paul2357/libraries_v2/HomeKit-ESP8266/src/storage.c:137: undefined reference to `_EEPROM_start'

/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /mnt/create-efs/webide/29/f3/29f32a690ec98f2e231ae59981834f97:paul2357/libraries_v2/HomeKit-ESP8266/src/storage.c:140: undefined reference to `_EEPROM_start'

collect2: error: ld returned 1 exit status

Error during build: exit status 1

Let me know if I need to provide any further info or if there is something that I can update/be updated to fix this!

Sobajdin, 2023-06-05:
Most probably you didnt defined proper size of flash for your board (e.g.) 4 Mb flash…

Paul2357, 2023-06-08:
Weird...I had it all setup with no issues. Tried compiling again, and no issues. Now I just can't get it to show up in my home app using the code I set up

closed 2023-06-12 by Paul2357