Mixiaoxiao/Arduino-HomeKit-ESP8266 #228: LDR Light Sensor

N5NMR, 2023-07-03:
Following the Example Code for Light Sensor with an LDR. Sensor shows value in Serial Monitor and is accepted by Apple Home as a new sensor BUT HomeKit says that this Sesor is not supported. 

Any ideas?

HomeKidd, 2023-09-06:
@N5NMR Can you post the code for reference?

N5NMR, 2023-09-08:
Sure, Here it is:

#include <Arduino.h>
#include <arduino_homekit_server.h>
#include "wifi_info.h"

const int ldrPin = A0;

#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

void setup() {
	Serial.begin(9600);
	wifi_connect(); // in wifi_info.h
	//homekit_storage_reset(); // to remove the previous HomeKit pairing storage when you first run this new HomeKit example
	my_homekit_setup();
}

void loop() {
	my_homekit_loop();
	delay(10);
}

//==============================
// HomeKit setup and loop
//==============================
extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_light;

void my_homekit_setup() {
	arduino_homekit_setup(&config);
}

static uint32_t next_heap_millis = 0;
static uint32_t next_report_millis = 0;

void my_homekit_loop() {
	arduino_homekit_loop();
	const uint32_t t = millis();
	if (t > next_report_millis) {
		// report sensor values every 10 seconds
		next_report_millis = t + 10 * 1000;
		my_homekit_report();
	}
	if (t > next_heap_millis) {
		// Show heap info every 5 seconds
		next_heap_millis = t + 5 * 1000;
		LOG_D("Free heap: %d, HomeKit clients: %d",
				ESP.getFreeHeap(), arduino_homekit_connected_clients_count());

	}
}

void my_homekit_report() {
	float ldrValue = analogRead(ldrPin); //int
  float licht = map(ldrValue, 0,1024,0,100);
  cha_light.value.float_value = licht;
	homekit_characteristic_notify(&cha_light, cha_light.value);
  LOG_D("Lichtlevel: %.1f", licht); 
}

HomeKidd, 2023-09-08:
This part of the code looks good, could you post the accessory (services, characteristic) setup too?

N5NMR, 2023-09-08:
Sure, Thanks

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

// format: float; min 0.0001, max 100000
homekit_characteristic_t cha_light = HOMEKIT_CHARACTERISTIC_(CURRENT_AMBIENT_LIGHT_LEVEL, 1);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {

    	HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Light Sensor"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Jann´s Bastelbude"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "23001"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266 LDR"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
        NULL
    }),
    NULL
};

 homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};