Mixiaoxiao/Arduino-HomeKit-ESP8266 #229: Sensors looses Connection

N5NMR, 2023-07-03:
Working with Wenos D1 Mini and SHT 30 Sensor. 

All Sensors looses connection to Apple HomeKit on a unregulated basis. 

I use only SHT 30 sensor, but on one I also use an I2C Display and on two others I use Voltmeter as Analog Display. 

Any suggestions?

Grwmagad, 2023-07-24:
Same problem here and probably the same as #213
I am trying to ESP.restart() after 0 homekit clients have been reported for a few times.
But thats obviously not the right way.

N5NMR, 2023-07-26:
> Same problem here and probably the same as #213
> 
> I am trying to ESP.restart() after 0 homekit clients have been reported for a few times.
> 
> But thats obviously not the right way.

Did this also. But now the Sensors are kicked out by apple and need to reinstalled… f@ck

Grwmagad, 2023-07-26:
I do not have issues with that.
I only added some EEPROM variables that erase my pairing but thats my fault. I will change the address where the EEPROM will be written and it will be fine.
ESP.restart is not resetting the pairing.
Are you sure that there is no homekit_storage_reset in the setup() function? Or somewhere else?

N5NMR, 2023-07-28:
Sure, all HomeKit storage reset has been out // in the code.

However. After implementing a reset in the code it Worked for a week or so and suddenly kicked out twice a day. No idea. Checked everything twice. 

Nerv wrecked

jockethebastard, 2024-04-25:
Any fix do this??

N5NMR, 2024-04-25:
The only thing that works ist to count the arduino_homekit_conecction_client_count == 0 (in my case till 25) and then do a reset.

Now it worked for a while. Anyway. Sometimes all sensors are kicked out completely by apple and need be reinstalled. No idea why

jockethebastard, 2024-04-26:
> The only thing that works ist to count the arduino_homekit_conecction_client_count == 0 (in my case till 25) and then do a reset.
> 
> Now it worked for a while. Anyway. Sometimes all sensors are kicked out completely by apple and need be reinstalled. No idea why
> 

I don't know exactly what you mean, but I'll try to figure it out. 
I set up a brand new ESP8266 yesterday with my temp sensor, and it worked right away, no problems adding it to HomeKit. But, since I have to move the sensor and disconnect it from the power source, it's then not working. Says it's not compatible and then never to be found again, unless I reinstall it.

N5NMR, 2024-04-26:
Sound like you have the HomeKit.reset still in the setup. // it after you upload the code. And upload again. It's stupid but works

I send the reset code later

N5NMR, 2024-11-23:
Main Importend hing i learn ist to put the esp on 160MHz.

That helps a Lot