Mixiaoxiao/Arduino-HomeKit-ESP8266 #232: New architecture

thncode, 2023-08-12:
Does anyone here know if this lib can work with HomeKits new architecture? I do have 19 different accessory types based on this very fine lib in use and would die if this stops working properly...

rayenking, 2023-08-17:
I have the same issue, I can't pair the homekit because new architecture :"

PimV5755, 2023-10-03:
I have problems since 16.7. After a powerdown of an accessory the homeapp lost is. It gives "no Reaction".
It appears out of the HomeKit database because if I look for new accessories my old one is there and I can re-pair it.
Very strange and very frustrative.

thncode, 2023-10-05:
OK - I did it. Had to re-configure all accessories but all works well now...