Mixiaoxiao/Arduino-HomeKit-ESP8266 #233: Compatibility issue with ios 16.6

doboti, 2023-08-18:
The whole peckage worked just fine, but when i downloaded the latest version of ios wich is actually 16.6 my lightbulbs just stopped working, and i couldn't pair them anymore. Tried on an other iphone with the same ios version, but same issue. I tried to upload an example code (example 1, 4, 5) but my phone never saw the accessories. If not the ios version what else could couse the issue?

vishal-codinghub, 2023-08-18:
I have iOS 16.6 and so far I tried following... 

Example 2 works quite reliably. I can control and see device in the Home App.
Example 3 Stateless Programmable Switch: lets me walk through Home app wizard until the location of device - Living room etc. At the end does not show in the App.

romaneightpointzero, 2023-08-21:
After the HomeKit update 16.6 and the new architecture HomeKit says Device unavailable.
I fixed the problem with a delete of the not working devices and new upload of the sketch with active homekit_storage_reset() - function.
After that I registered the device as a new device via HomeKit-App.
Uploaded the sketch with deactivated reset - Function.
Now it works again.

NS-K, 2023-09-08:
Can you please elaborate?
What do you mean by deactivated reset function

Grwmagad, 2023-09-08:
homekit_storage_reset() <- reset function

just deactivate it by adding comment "//" in front of the line

romaneightpointzero, 2023-09-08:
> Can you please elaborate?
> What do you mean by deactivated reset function

> homekit_storage_reset() <- reset function
> 
> just deactivate it by adding comment "//" in front of the line

Already answered.

ZOP9974, 2024-11-10:
Is iOS 17.7 incompatible? Why does frequent display not affect
ios17.7不兼容吗？为什么很频繁的显示未影响