Mixiaoxiao/Arduino-HomeKit-ESP8266 #234: Use Homekit-ESP8266 to control a stepper driver.

robban1113, 2023-08-23:
Hi guys! First of all I want to say thank for a great library(even though I don't master it myself yet)!

I'm renovating my apartment but now everything stands still due to an issue.

The think is I'm new into Arduino and programming and have tried to make a smart device to my home.
I know that there is plenty of options that makes it possible to control a stepper motor thru wifi but I want to be able to integrate it to my Homekit app without having to use Homebridge.

In this project I use Wemos D1 mini, ULN2003 driver and 28BYJ-48 stepper motor to control my blinds.
Early on I tried to use the "simplest led" example to check if I could control the d1 mini which works great and I like how easy it is to integrate it to my Homekit app. 

The issue for me is that I don't know how to make it control my stepper motor instead of making the built in LED.
Could someone please provide me with the code so that I can make this possible? Please!

robban1113, 2023-08-23:
Created a repository if that helps.
[https://github.com/robban1113/Blinds](url)

robban1113, 2023-08-24:
Updated the accessory category in the .C file.

theolecoutre, 2023-11-08:
Hello, have you tried using this [library by maximkulkin](https://github.com/maximkulkin/esp-homekit) ?

I'm using this library with an ESP8266, an ULN2003 and a 28BYJ-48 as you, and it's working pretty well.