Mixiaoxiao/Arduino-HomeKit-ESP8266 #239: hiding accessories

thncode, 2023-11-03:
I could'nt find that by searching and experimenting - is it possible to hide accessories at run time?

d4rkmen, 2023-11-03:
If you change config number value in txt record, the controller will reload accessory information.
So, remove accessory or some service from the description, and shift config number

thncode, 2023-11-03:
Sorry, what is 'config number value in txt record'? Could you please paste a code snippet?

arturlukosek, 2023-12-31:
@thncode have you managed to figure out how to hide accessories?

thncode, 2024-01-02:
no - I did'nt understand the comment of @d4rkmen. I had to make a workaround by using two sets of accessories, which of course does not work for all use cases...

maccoylton, 2024-01-03:
HomeKit publishes via mdns some info that helps the client app, for example the Apple Home app, discover and use your accessory. Within that mdns record there is information including a config number. I think the suggestion is if you increment that config number then the client will pickup any changes, seem to remember that is mentions in the homekit spec. 

It is stored within the struct you create with homekit_server_init.

d4rkmen, 2024-01-03:
Controller will reload accessory information when config number changes. This is the way it works