Mixiaoxiao/Arduino-HomeKit-ESP8266 #240: Disconnect on pair setup step 3/3

ciananDR, 2023-11-13:
**Below is my serial monitor:**

>>> [  20226] HomeKit: Got new client: local 192.168.0.131:5556, remote 192.168.0.130:52667
>>> [  20236] HomeKit: [Client 1073685052] Pair Setup Step 1/3
>>> [  22870] HomeKit: [Client 1073685052] Pair Setup Step 2/3
>>> [  22884] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  42140] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Current temperature: nan
Current Humidity: nan
Free heap: 38232, HomeKit clients: 1
>>> [  51871] HomeKit: [Client 1073685052] Pair Setup Step 3/3
>>> [  51914] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  53339] HomeKit: Added pairing with 378440F9-2E48-4004-87C7-3573BF148FA6
>>> [  53441] HomeKit: Free saved_preinit_pairing_context
>>> [  53446] HomeKit: [Client 1073685052] Successfully paired
>>> [  53585] HomeKit: [Client 1073685052] Disconnected!
>>> [  53590] HomeKit: [Client 1073685052] Closing client connection
>>> [  53607] HomeKit: Got new client: local 192.168.0.131:5556, remote 192.168.0.130:52670
>>> [  53617] HomeKit: [Client 1073679460] Pair Verify Step 1/2
>>> [  54107] HomeKit: Free heap: 40352
>>> [  54143] HomeKit: [Client 1073679460] Pair Verify Step 2/2
>>> [  54150] HomeKit: [Client 1073679460] Found pairing with 378440F9-2E48-4004-87C7-3573BF148FA6
>>> [  54180] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  55622] HomeKit: [Client 1073679460] Verification successful, secure session established
>>> [  55630] HomeKit: Free heap: 40464
>>> [  55706] HomeKit: [Client 1073679460] Get Accessories
>>> [  55929] HomeKit: [Client 1073679460] Disconnected!
>>> [  55935] HomeKit: [Client 1073679460] Closing client connection
Free heap: 42584, HomeKit clients: 0
>>> [  59237] HomeKit: Got new client: local 192.168.0.131:5556, remote 192.168.0.130:52671
>>> [  59258] HomeKit: [Client 1073679460] Pair Verify Step 1/2
>>> [  59749] HomeKit: Free heap: 40208
>>> [  59783] HomeKit: [Client 1073679460] Disconnected!
>>> [  59789] HomeKit: [Client 1073679460] Closing client connection

lumenhu, 2023-11-17:
me too, it worked very well two days ago, but I encounter this problem today.

blazing1991, 2023-11-26:
Exactly same here, I'm on latest iOS 17.1.1

yvesgarnier, 2023-12-06:
i have the same issue. have you already found the solution? could it be because it is not possible to enter the correct password format? 111-11-111 instead of 1111-1111?

blazing1991, 2023-12-07:
> Exactly same here, I'm on latest iOS 17.1.1

I also resolved the issue, in my case the problem was in a defective DHT11 sensor, for some reason it was blocking the ESP from finishing setup. 
First I uploaded a clean example of a temperature sensor and it worked flawlessly. Then I replaced the DHT11 sensor and tested it with a simple sketch to make sure it was giving appropriate numbers. Then combined the code and successfully added esp8266 into the homekit. 
BUT this piece of sh*t didn't want to work properly anyways, each 1-4h it just loses connection and I had to restart it, which is not what I expected. 

After fighting with it for a week I just ended up switching to ESP32 and ESP32-HomeSpan library, which has been working without any problem for a week now.