Mixiaoxiao/Arduino-HomeKit-ESP8266 #244: Implement temperature, humidity and 2 switches and oled 0.96 to monitor the sensors.

KaitoKK1412, 2024-01-30:
Can anyone help me to add two switches and an oled 0.96" so that it shows the temp and humidity in the first sample temperature, humidity. I am new to coding and want to learn new things. I would be thankful if anyone can help me.

stuntstein, 2024-03-25:
Just the other day I made this little project
https://github.com/stuntstein/Arduino_homekit_temp-humi-sensor

jockethebastard, 2024-03-25:
> Just the other day I made this little project https://github.com/stuntstein/Arduino_homekit_temp-humi-sensor

Cool! I’ll have to check it out, since I can’t get the temp working with any other code, and I’ve tried so many different versions. I also have an oled display never used.

stuntstein, 2024-03-25:
[image]

jockethebastard, 2024-03-25:
Looks awesome! I guess it’s compatible with smaller oled screen, by adjusting the pixels right?

stuntstein, 2024-03-25:
This is one of the smallest, but yes you can change the resolution:
```
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 32 // OLED display height, in pixels
#define SCREEN_ADDRESS 0x3C ///< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32
#define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
```

You would also have to adapt the x,y offset when printing your text
```
  display.setCursor(0, 12);
  sprintf(s,"Tmp: %5.1fC", temp);
  display.print(s);
```

sbum8181, 2025-01-04:
accessory is out of comliance 

log is here-------------------------------------------------------------

[ 11472] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
[ 14783] HomeKit: Preinit pairing context success
[ 14788] HomeKit: Configuring MDNS
[ 14794] HomeKit: MDNS begin: Temperature Sensor-D, IP: 192.168.29.25
[ 14801] HomeKit: Init server over
Current temperature: nan
Current humidity: nan
Free heap: 40384, HomeKit clients: 0
Free heap: 40280, HomeKit clients: 0
[ 11505] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
[ 14838] HomeKit: Preinit pairing context success
[ 14843] HomeKit: Configuring MDNS
[ 14848] HomeKit: MDNS begin: Temperature Sensor-D, IP: 192.168.29.25
[ 14855] HomeKit: Init server over
Current temperature: 18.0
Current humidity: 78.6
Free heap: 40160, HomeKit clients: 0
Free heap: 40160, HomeKit clients: 0
Current temperature: 18.0
Current humidity: 78.7
Free heap: 39408, HomeKit clients: 0
Free heap: 39992, HomeKit clients: 0
[ 31344] HomeKit: Got new client: local 192.168.29.25:5556, remote 192.168.29.132:51611
[ 31354] HomeKit: [Client 1073683340] Pair Setup Step 1/3
[ 34842] HomeKit: [Client 1073683340] Pair Setup Step 2/3
[ 34853] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
[ 44320] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Current temperature: 18.0
Current humidity: 78.8
[ 34853] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
[ 44320] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Current temperature: 18.0
Current humidity: 78.8
Free heap: 38224, HomeKit clients: 1
[ 49114] HomeKit: [Client 1073683340] Pair Setup Step 3/3
[ 49141] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
[ 49858] HomeKit: Added pairing with 486BE470-518D-4553-8739-EC1330E87380
[ 49923] HomeKit: Free saved_preinit_pairing_context
[ 49929] HomeKit: [Client 1073683340] Successfully paired
[ 50178] HomeKit: [Client 1073683340] Disconnected!
[ 50183] HomeKit: [Client 1073683340] Closing client connection
[ 50200] HomeKit: Got new client: local 192.168.29.25:5556, remote 192.168.29.132:51612
[ 50210] HomeKit: [Client 1073682876] Pair Verify Step 1/2
[ 50471] HomeKit: Free heap: 40160
[ 50536] HomeKit: [Client 1073682876] Pair Verify Step 2/2
[ 50543] HomeKit: [Client 1073682876] Found pairing with 486BE470-518D-4553-8739-EC1330E87380
[ 50564] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
[ 51291] HomeKit: [Client 1073682876] Verification successful, secure session established
[ 51300] HomeKit: Free heap: 40272
[ 51447] HomeKit: [Client 1073682876] Get Accessories
[ 51646] HomeKit: [Client 1073682876] Disconnected!
[ 51651] HomeKit: [Client 1073682876] Closing client connection
Free heap: 42680, HomeKit clients: 0
Current temperature: 17.9
Current humidity: 78.9
Free heap: 42680, HomeKit clients: 0
Free heap: 42680, HomeKit clients: 0